MEGA65/mega65-core #1: SD Card reader issues

gurcei, 2016-08-08:
I'm presently having trouble getting the kernel read from the sd-card and into memory.

There seems to be a chunk of incorrect data in the data, described in this forum post:

https://groups.google.com/d/msg/c65gs-development/q7kCars2Xqw/a_WIWMmcAgAJ

Paul has a few suggestions to chase up that he mentions in that forum thread.

Ben-401, 2016-08-10:
Paul suggests this is one of my initial tasks.
also this thread:
https://groups.google.com/forum/#!topic/c65gs-development/lfTTbbDhWgo

gurcei, 2016-08-10:
No worries and fair enough too, as it's a fairly nasty issue that needs some heavy focus, and I'm getting fairly distracted with m65dbg as is. I'll still keep one eye on looking into this too, as it provided the impetus for the growth of the debugger, so I might still chip in my 10 cents worth on this topic from time-to-time if I notice anything.

gurcei, 2016-08-16:
While debugging tonight, discovered a potential cause of the problem. More info in this forum-post:

https://groups.google.com/d/msg/c65gs-development/lfTTbbDhWgo/5Fm-_td2AwAJ

gardners, 2016-08-16:
How about (untested):

dos_file_read_current_sector:
jsr dos_set_current_cluster_from_file
jsr dos_cluster_to_sector

; Add sector within cluster
jsr dos_get_file_descriptor_offset
bcc dos_return_error_already_set
ora #dos_filedescriptor_offset_sectorincluster
tay
ldx #$00
clc
php
lda dos_file_descriptors,y
bne dfrcs4
dfrcs2: lda #$00
dfrcs4: plp
adc $d681,x
sta $d681,x
bcc dfrcs_inced
php
inx
cpx #$04
bne dfrcs2
plp
dfrcs_inced:

i.e., save and restore the processor flags. Cost is 4 bytes and some cycles.

Do check that what I have suggested actually does the job before committing
;)

Paul.

gurcei, 2016-08-19:
Yep, Paul's suggestion solved the prob and he's pushed in the fix into the master branch.

There's another issue regarding the failed loading of the (optional) KICKUP.M65 file from the sd-card, I'm looking into that next.

gurcei, 2016-08-22:
I've tested with the latest bit-stream (I was on an older bitstream previously), and it was able to load the KICKUP.M65 successfully. Ok to close? Or other known quirks of behaviour in the sd-card reading that still remain?

gardners, 2016-08-22:
Hello,

If there are no known outstanding items, close the request. We can always
reopen it or open a new one if (or when ;) we find new problems.

Paul.

Ben-401, 2016-08-26:
i have confirmed that the SDcard reader sometimes needs to call the "redoread" function.
it seems to do this when it detects that it did not read 2x-pages (512 bytes).
in this regard i reverted the debug message to display this abnormality, in an attempt to characterise the behaviour.

It seems to consistently re-do-read when loading bitstream from USB and having KICKUP on SD.

Ben-401, 2016-08-29:
I have reviewed the "kickstart" and "dos" code, and have modified the Checkpoint messages that are generated. I now have a better understanding of how it works.

I have also created a script that examines a sdcard and writes information to harddisk.
I have also created a script to zero/partition/format a sdcard in an attempt to try diagnose the problem.

I will upload these files to the repo soon, in the src/utilities directory.

Ben-401, 2016-08-31:
i have found some unusual behaviour in the _dos code:
- re-reading MBR (sector 0) when not neccessary
- checking for $55AA when not nessessary
- not observing file attributes appropriately.
- seems to be using CHS (cylinder/head/sector) i think
- a heap of code for file handlers and offset information...

I have modified the _dos code, available for viewing on MVFILE branch
but is under development, please do not merge or branch from.

gardners, 2016-08-31:
Hello,

It should only be using LBA, not CHS addressing.

The file handler code etc is important, to allow a program to ask the
hypervisor to open multiple files at the same time.  However, we do know
that we have a file handle leak somewhere.

Paul.

Ben-401, 2017-02-14:
how do we test that this change infact works?
is there as basic program or assembly code that we can include in our test procedure?
Please... :)

And can this patch be applied to the development branch instead of cursorkeys?
I thought cursorkeys has been retired?

gardners, 2017-02-14:
Hello,

Falk and I are still pushing the cursorkeys branch forward as the focus of
our attention, as we are the only folks working on it.  It would be good to
merge all the changes onto development, of course.

Paul.

lydon42, 2022-01-09:
As reading and writing of the SD Card obviously works in 2022, we can close this.

closed 2022-01-09 by lydon42