MEGA65/mega65-core #2: Optimization of Serial Data

Kirb0031, 2016-08-09:
Regarding MEGA65/m65dbg#3 and serial speed issues. 

Current serial output is inefficient (repeats banner every time, includes spaces, etc).
A more efficient, but less human readable, protocol for use with m65dbg was suggested
which should improve debugging performance.

Ben-401, 2016-08-10:
maybe improve the existing format by not repeating banner etc, but keep this human-readable-format for users that do not want to use the m65dbg.
the fpga-side serial-debugger could by default use the above human-readible-format, and when it recieves a special message from the m65dbg, the fpga-serial-debugger could switch its format to the optimised one.

gardners, 2016-08-10:
This is the general plan: we can implement a command that switches the
mode.  That said, I am also comfortable with us making m65dbg a mandatory
tool for using the serial debugger, if it results in significantly improved
utility.

lydon42, 2024-08-17:
This was not worked on for 8 years, so I think we close it for now. If the need arises a new issue should be created.

closed 2024-08-17 by lydon42