MEGA65/mega65-core #4: update User-Manual from docs.google (practice your markdown)

Ben-401, 2016-08-10:
usermanual was sourced from
https://docs.google.com/document/d/1fmEUg6hDdWRb2tFZ3n4LG7S1mNP04_SUAW5DrE8zRpk/edit and now will be maintained at "/doc/usermanual0.md".
To do includes:  
1. add CR/LF, formatting etc
1. make it appear as close as possible to the original docs.google file.

gurcei, 2016-08-22:
I'd be happy to look into this one next. I reckon I'm due for a documentation task, just to let all this new knowledge in my brain solidify a little.

I couldn't figure out how to set the "Assignee" field via the github interface though.

gardners, 2016-08-22:
Assigned.

gardners, 2016-08-23:
Whoops. Assigned wrong person. However, if you both wish to work on it, that's fine by me, too.

gurcei, 2016-08-24:
Just a small note, I've installed grip and am now taking a look into the usermanual0.md page.

First point of interest is the initial table-of-contents section. Sadly, github's wiki markdown doesn't have any in-built method for doing an auto-generated TOC, so you have to generate it manually. Seems like it's a much-requested feature in this thread, but no movement on it as yet:

https://github.com/isaacs/github/issues/215

A few users seemed to have rolled their own methods to auto-generate tocs, I'll have a read of those next. Still, part of me wonders if github's wiki markdown language aims for a level of minimalism, hence it will never be as feature-rich as a stand-alone wiki.

Well, I'll make do for now, but still thought I'd better give it a mention.

Ben-401, 2016-08-24:
You can check my doc/structural.md page for example hyperlink TOC, and googlegroup explanation.

isaacs/github#215

gardners, 2016-08-27:
This is already looking very nice.

Please update the documentation to use MEGA65 instead of C65GS in most places.

Similarly, please change the commentary about C65 bitplanes: The plan is to implement them (and this is in fact mostly done).

gardners, 2018-02-08:
There is some recent work on creating per-topic .md files in the doc/ directory.  This might be the best way to approach this problem, and have TOC just as a file of links to the various files.  Separate files allows much easier parallel work on them, too.

lydon42, 2022-01-09:
Is this still an issue for mega65-core? Or was this solved by creating mega65-user-manual?

gurcei, 2022-05-30:
Feel like we can close this one. It relates to a time when details from a google-doc were migrated into this repo's .md files. I feel like other documentation practices are having more traction these days (user manual, meg65-book, filehost articles, etc), so think it's time to close this one.

closed 2022-05-30 by gurcei