MEGA65/mega65-core #7: serial buffer for in-built monitor

gurcei, 2016-08-10:
This one relates to #2 somewhat. Paul suggested the creation of a serial buffer for the comms sent out of the hardware by the monitor:

https://groups.google.com/d/msg/c65gs-development/FESl92s4vEM/rpmUe5Z4AQAJ

Perhaps this might improve performance too?

gardners, 2016-08-10:
The only performance boost it will give, is that you will be able to write
a whole command and flush at once.  USB latency is still likely to be a
remaining problem.

Ben-401, 2016-08-10:
i think the serial-comms needs to be a blocking-call. Ie your calling component waits until the message has been sent, then the calling component goes away and continues.
i dont think a buffer here will help at all, infact it will only duplicate the message, whereas my understanding is that the current architecture calling component sends a byte-at-a-time to the serial-monitor to transmit.
as for the serial-comms component, i think an improvement would be that the 2000-odd clock cycle delay be done-away with, and more tightly-timed circuit be implemented. I can elaborate on this. Its all about handshaking.

gardners, 2020-12-18:
The serial monitor is basically UART fed, so the lack of a UART RX buffer does mean that the sending side has to leave a long enough gap between each character for the monitor to process it. As the new monitor is actually running in a little CPU, this delay can be quite big. As a result we have to send each character with a considerable delay between them. Worse, because its over USB we have to make sure that the USB layer doesn't merge any of the writes into a single packet.  To make this stable, we typically have delays of 500 -- 2000 usec between each character, often more.  This means we effectively write at ~1KB/sec instead of the ~200KB/sec that the 2mbit/sec serial interface can support.  Thus just being able to write whole commands at once will likely result in substantially faster performance. But more importantly, it will greatly reduce the chance of characters being dropped, which does seem to be a problem when using Windows as the sender.

closed 2022-10-02 by lydon42