MEGA65/mega65-core #10: Add ability to when hypervisor runs code outside of the hypervisor ROM

gardners, 2016-08-12:
Typically if there is a bug in the hypervisor, it will end up running code outside of its $8000-$BFFF 16KB memory.  It would be helpful to have an option to trap when this occurs unexpectedly, so that hypervisor bugs can be tracked down more easily.

gardners, 2016-09-16:
This has been implemented in lgb's emulator, but not yet in the real hardware.  We still want it in the hardware, so that we can prevent various kinds of attempts to subvert the hypervisor.

lydon42, 2024-08-17:
@gardners is this something that is still unimplemented and that we might want?

gardners, 2024-08-17:
Yes, leave this hanging around, as I will need to look at it for the MEGAphone at some point.