MEGA65/mega65-core #17: Implement checking reading/writing offsets by hypervisor DOS against parititon boundaries

lgblgblgb, 2016-08-31:
# The problem

There can be cases when hypervisor DOS gets wrong about reading (or writing) from insane position, including:

* bug in the hypervisor DOS itself
* the file system is corrupted / contains errors what user uses
* the file system is valid FAT32, though hypervisor DOS can't use properly (expects different FS parameters, whatever)
* user program itself tries to read (or write) directly at block (not FS) level, where the given position is invalid

In any of the cases, it would be nice, to have some kind of "hypervisor DOS panic" (maybe with some debug info?) on the screen (even if it's not M65's fault, that the file system is bad). Currently, as far as I can see with my M65 emulator tries (Xemu), in case of invalid operation offset, hypervisor just gets back an error, and it re-tries again and again, and the user has no idea what happening in the background. I have the problem with Xemu, that with my (otherwise seems to be valid) FAT32 file system caused DOS tried to read something well beyond the total size of the SD card actually, and it stuck in a loop to try again and again. In Xemu, I could handle that with doing offset check for the whole emulated card image and panic'ing by Xemu itself, but it should be nice to have some kind of similar feature built-in.

# Possible solution

I think, it would be nice, to create a table (if not done already) of used partitions, about the "offset range" which is valid for the given partition. On each read/write ops, it should be checked that the requested offset is in the range. If it's not, there can be that "panic" stuff.

lgblgblgb, 2016-08-31:
Other than the partition based check, even just a single check at ever op for the the total device size can help too. The partition "range" check is a more sophisticated version.

lydon42, 2024-08-17:
HYPPO dos_cluster_to_sector still has a comment at the end stating:
```
        ;; XXX - Check that result does not exceed fs_start_sector+fs_sector_count
        ;; and run over into another partition
```
so I suppose we don#t check this yet.

gardners, 2024-08-17:
Yes, we don't check this yet. If I implement the hardware FAT acceleration #766 , it will come for free.