MEGA65/mega65-core #21: Possible solution: KS crashes if no MEGA65.D81 could found during the boot?

lgblgblgb, 2016-09-15:
At least for me, if there is no `MEGA65.D81` on the SD card, KS causes to "out of hypervisor memory" execution. As far as I can tell this caused a BCC op in `kickstart.a65`, line 649 which would branch to label `nod81` if no file found. However since this is the main execution "flow" of booting with empty stack, and nod81 is a routine having RTS on the end, it seems this was, what caused the stack "underflow" and execution at an insane address popped from the stack by RTS.

Though I am not really sure about the kickstart structure yet, I think, `BCC` mentioned above should be modified to branch to `d81attachfail` instead of `nod81`. I've tested that modification of mine, and it seems KS booted normally with M65 emulator, using my "faulty" FAT32 card image, with no MEGA65.D81 (but ROM is OK for sure). It was tested with the EOC/dirent-end patches combined, and it seems with this fix as well, everything seems to be normal.

lgblgblgb, 2016-09-15:
In addition of my previously not working FAT32 fs, I tried with the "official" image (what I got from Deft) and it seems it also works, with my M65 emulator I could boot, go into 64 mode running WGM and starting synthmark 64.

gardners, 2016-09-15:
Possible fix to missing mega65.d81 induced crash in commit db64668

gardners, 2016-09-16:
@lgblgblgb I've assigned you, so that if you can confirm that this latest patch has fixed it, you can close the bug.

lgblgblgb, 2016-09-18:
Indeed, it fixed the issue for me! Thanks. Closing ...

closed 2016-09-18 by lgblgblgb