MEGA65/mega65-core #22: FDC emulation doesn't work for writing sectors

gardners, 2016-09-16:
There seem to be two problems here:

1. C65 ROM gets stuck waiting for busy flag

2. The sector gets written with the data rotated by 2 bytes.

gardners, 2016-09-16:
The out-by-2 error could be caused by the F011 write sector emulation copying the FDC sector buffer to the SD card's sector buffer.

The f011_busy flag seems that it should get cleared in the appropriate places, so that's a bit of a mystery.

lgblgblgb, 2016-09-16:
We should know the exact point in C65 ROM where it is while it's waiting. I **think** that busy flag is more a controller internal thing to signal "processing command". The actual read or write (from/to disk, to/from FDC buffer) is signalled by RDREQ and WTREQ. However I can be only sure with testing a real C65 on this. In my emulator actually I never set BUSY :) Though in case of read, RDREQ is a must, since DOS code waits for it. Not sure about WTREQ though. For the out-by-2 problem: maybe it should be tested in the VHDL implementation that it's off-by-2 already in the FDC buffer. Anyway even if it's not correct literally, since FDC can not write the "part" of a physical sector (or read) every operation should start from buffer position 0, when read from disk or write to disk (the CPU's pointer is another question). So you can simply use the FDC buffer as-as always from beginning (pos zero) when write to the SD card? Just ideas from me ...

gardners, 2016-09-16:
The busy flag is important if the read/write takes non-zero time from perspective of the CPU, so that it doesn't send another disk operation while the last one is still working.  I can already identify the location in the ROM where it is stuck waiting for BUSY to clear.

Some further examination shows that I never really finished implementing writing, and it is a complete mystery how the data even gets to the write buffer at all.  The good news, is that this should be quite easy to run under simulation to work out how the buffer is getting copied and rotated in the process.

lgblgblgb, 2016-09-16:
OK, you're right, I meant more about reading when C65 DOS ROM code seems to wait only (after the READ command) for the high level of RDREQ to start read FDC buffer (which checking EQ bit before each read), and nothing about BUSY too much as far as I can remember now (BUSY in this case is only checked before issuing the READ command itself to see, if FDC is "idle" at all to be able to deal with a new command). That is, even for READ command there can be a problem if "F011 emulation" if it's too fast, giving only a "spike" in RDREQ signal, the CPU may miss then ("already gone"), thus hanging even in case of read in DOS ROM code ... This can be even the reason of some EQ anomaly, that normally things just work, because the typical speed of F011, and floppy disk drives in general are okey for the CPU not to miss some signal transits ... Without that (like with my primitive approach of F011 emulation) custom workarounds are needed ........

gardners, 2019-04-24:
Long ago fixed.

closed 2019-04-24 by gardners