MEGA65/mega65-core #23: Ask for implementation of the hypervisor debug UART ready bit

lgblgblgb, 2016-09-18:
Dumping CPs (checkpoints) by KS is a nice way to have some information about hypervisor DOS and KS in general. However, on a real M65 it's a serial port after all, thus there is some delay need to be provided after every characters written out, which is understandable. Currently there is some "busy loop" solution to "burn" enough CPU cycles to provide the needed delay. However for the M65 emulator it's quite uncomfortable since tons of unneeded information will be provided with its own debugging (even at per opcode basis) methods, because of the execution of the delay loop. Also, in the future, some may expect faster CPU implementation for the M65, so it would be better for all, not to depend on "manual timing" but on an implemented "ready bit" by the M65 itself. So, basically the hardware would know if it's ready. On the M65 emulator, I would fake "always ready" as there is no real CPU time needed to be ready there.

gardners, 2016-09-19:
Agreed, and we will implement this.

Ben-401, 2017-03-20:
Yes, this has now been implemented.
Refer:
https://github.com/MEGA65/mega65-core/commit/f93ca46938096cad0387ae78d5274b881e9a51d3
and
https://groups.google.com/forum/#!topic/c65gs-development/84G_DDVFhqM
Has been tested on hardware, and required Kickstart(KS) assembly (ASM) to be modified to "before sending a char, wait for the uart to be ready {infact NOT-BUSY}.

This is implemented in KS by:
+	ldx #$00
+cp5:
+	; wait for uart to be not busy
+	lda hypervisor_write_char_to_serial_monitor	; LSB is busy status
+	bne cp5		; branch if busy (LSB=1)
+	; uart is not busy, so write the char
+	lda msg_checkpoint,x
+	sta hypervisor_write_char_to_serial_monitor
+	inx
+	if more chars to send, then bne cp5

The hardware runs much faster now due to minimising the delay to only what it needs for the baud-rate.
The XEMU/mega65 runs much faster now doe to not having to do the 2k delay loop.

closed 2017-03-20 by Ben-401