MEGA65/mega65-core #26: Consider updating soft-SID voice to this implementation

gardners, 2016-11-10:
Is this one better? Is the current one better?
Someone will have to try it to find out.

[m6581_voice.zip](https://github.com/MEGA65/mega65-core/files/583653/m6581_voice.zip)

gurcei, 2018-01-19:
I was pondering looking into this (I might be biting off more than I can chew, let's see). I was curious if this would improve the sound in "wizard of wor", as it sounded quite off the last time I played it (with the present sid implementation).

I noticed that the previously attached 'm6581_voice.vhd' file came from the fpga arcade project (via the comments within). It also refers to just a single voice of the sid chip, I located the full vhd here:

http://svn.fpgaarcade.com/viewvc/ReplayPub?view=revision&revision=99

But I also noticed that the author removed all c64-related vhdl from the repository in revision 113 (on 29/10/2016):

http://svn.fpgaarcade.com/viewvc/ReplayPub?view=revision&revision=113

I thought he might have located the files elsewhere within the directory structure, but after browsing the latest repository contents, they are no-where to be found.

Ok, so I guess it'll be best to reference the repository as it was in rev112 (just prior to the deletion):

http://svn.fpgaarcade.com/viewvc/ReplayPub/hw/replay/cores/c64/?pathrev=112

I was tempted to find out why he deleted them for... I might ask on the forums for that project...

gurcei, 2018-01-19:
Just doing some detective-work on the fpgaarcade forum. In one thread [here](http://www.fpgaarcade.com/punbb/viewtopic.php?id=1201), there was a post from WoS (dated 29/10/2016) where he mentioned there being a private svn repo for developers on the project and a public svn repo where certain contents from the private repo are cherry-picked across to the public repo for mirroring at midnight every day.

He also said:

> I didn't add the source trees, as I don't know if the individual contributors want to have them released.

So perhaps this is the reason why... I'll confirm with a post on their forums once I get registered.

gurcei, 2018-01-19:
Got a reply from Mike on the forums. They're undergoing a re-write at the moment, after this, they'll make their sources public again, nice :)

Till then, I'm happy to try out the vhdl as it was at rev 112 (prior to 29/10/2016). I compared the in and outs of the existing "sid_6581.vhd" again their "m6581.vhd", seems mostly the same, but a few small discrepancies that I can't fathom yet, and I've run out of steam for tonight.

```
               sid6581                                       m6581                   
              ---------                                    ----------                
  clk_1MHz  ->|       |                           i_clk -> |        |                
     clk32  ->|       |                           i_ena -> |        |                
     reset  ->|       | -> audio_data[18]         i_rst -> |        | -> o_audio[16] 
        cs  ->|       |                       i_addr[5] -> |        | -> o_audio_we  
        we  ->|       |                          i_rw_l -> |        |                
    addr[5] ->|       |                          i_cs_l -> |        |                
      di[8] ->|       | -> do[8]              i_data[8] -> |        | -> o_data[8]   
   pot_x[8] ->|       |                       i_padx[8] -> |        | -> o_data_oe   
   pot_y[8] ->|       |                       i_pady[8] -> |        |                
              ---------                                    ----------                
```

gardners, 2018-01-19:
Hello,

I'm inclined to leave it until they rerelease the code, so that we don't
have to repeat the work.  There will be some changes that I made to
interface to the 50MHz IO bus off the MEGA65.

Paul.

gurcei, 2018-01-20:
Fair enough, I'll leave this one on the backburner for now.

deftmega, 2022-01-03:
project is dead, was most probably "bought out" by a well known person. We continue our own way :)
Lydon to close.

closed 2022-01-04 by lydon42