MEGA65/mega65-core #31: Misleading error message when SD card is unreadable

gardners, 2017-03-19:
When an SD card cannot be read for some reason, it usually shows up as a "00 partitions mounted", instead of saying something like "cannot read master boot record", which would be more informative.

rdpeake, 2021-02-05:
simple reproduction via using an sd card without a mbr in a mega65

lydon42, 2021-12-25:
Tested on nexysA7 with development@e485611 build 147 with both a zeroed and a randomly filled 8GB sdcard (dd to sd device, killing the mbr):
With both HYPPO will display
`DISK-COUNT=00, DEFAULT-DISK=00`
and then ask which utility I like to start. Selecting 2 gives me the sd format utility which will detect the unformatted sd and ask me if I like to partition & format it.

@rdpeake do I need to do something else with the card to trigger the problem? Otherwise it seems to have been fixed.

closed 2022-10-02 by lydon42