MEGA65/mega65-core #33: Sprite X-position issue in H640 mode

lgblgblgb, 2017-04-04:
In my "wannabe CP/M emulator" for M65 I discovered the issue, that sprites' horizontal coordinate is not there where it should be. I can be wrong, but AFAIK, sprite position etc should be independent of the used video mode. So I used the same theory to calculate the X coordinate as it would be on C64 (or so ...) but the sprite then is far too be right compared to where it should be. You can see the issue in this video (around 0:35), the "cursor" I use there is a sprite for real, and I calculate its position from the text coordinates:

https://www.youtube.com/watch?v=iJwvp6lZ2DU

frehwagen, 2017-04-24:
This might be addressed with the pull request https://github.com/MEGA65/mega65-core/pull/34

rdpeake, 2021-02-05:
fixed via `H640 flag for sprites to change sprite resolution`

closed 2021-02-05 by rdpeake