MEGA65/mega65-core #35: Possible problem with hypervisor if VIC-III style ROM mapping is active

lgblgblgb, 2017-04-22:
Sorry if I over-looked something, but by inspecting the VHDL code (to develop Xemu further with better compatibility) I have the suspect (but I can be easily wrong here!), that there is an issue if a hypervisor trap is taken while VIC-III ROM mapping is active, especially for ROM@8000 bit. Since - AFAIK - VIC-III ROM mapping is "stronger" than MAP, mapping the hypervisor memory from $8000 (done during the process entering into hypervisor mode) will be overridden by this ROM mapping, so there is a potential crash after this. Maybe other ROM mapping bits can be problematic for other reasons as well (like passing parameters in memory for hypervisor DOS calls, or whatever). I think, VIC-III register $30 should be saved on entering hypervisor, and ROM mapping bits should be zeroed. On leaving hypervisor, of course, reg $30 should be restored. Probably the same way, as other things (like original MAP state) are saved and restored.

gardners, 2017-04-23:
Hello,

Thanks -- I'll make a change to make sure that the VIC-III ROM mappings
don't work in hypervisor mode.

Paul.

gardners, 2017-04-23:
Hello,

The change has been made on the m65pcb-ports branch.  It might be a while
before this percolates back through to development and master.

Paul.

lydon42, 2024-08-17:
I have to assume this is no longer an issue. The branch mentioned by @gardners does not exists anymore, and the "stronger" mapping of VIC-III mentioned by @lgblgblgb also is not true anymore by our latest discoveries: MAP is always stronger.
So I will close this, please file a new issue if this is an issue, but take in account the changed situation.

closed 2024-08-17 by lydon42