MEGA65/mega65-core #36: New memory mapping whish(es)

lgblgblgb, 2017-04-28:
Somewhat complex topic ....

1. There is a C65-M65 incompatibility issue that is, the last 2K of chip-RAM is the first 2K of colour RAM. On C65 this area can be used for VIC-III to fetch graphics data (eg bitplane) too if some doesn't need the colour RAM functionality. AFAIK it wouldn't work with M65. I would suggest to use 128K chip RAM (and not 126) by VIC-IV anyway on M65. The last 2K would be "overlapped" by the colour RAM 2K though, CPU would see that all the time, however VIC-IV for non-colour fetch purposes would always see the chip RAM there too. Thus, only the policy is needed that writing that area would affect both of colour RAM and chip RAM at the same time. That would solve this compatibility problem, as - I guess - for timing issues it wouldn't too much possible other way.

2. For some purposes, it's kinda uncomfortable to always have the CPU port at phys addr 0/1 and the colour RAM at the end of 128K. What I would love, is to map 256K RAM (chip+fast) without the overlapping colour RAM 2K + I/O port at another, higher physical address, so it would be possible for M65-aware programs to MAP it, linear-opcode-addressing it, or DMAing it if needed. It's only really useful though, if chip the colour RAM and the CPU port is not there. This also means, that care must be taken: if some write the last 2K of chip RAM at this higher mapping address, there would be a difference between colour RAM and chip RAM "write through" policy (see point 1) so it should be handled with care. But it's an M65 feature, so only M65-aware things could do that for real. I would vote not to take account (but surely it doesn't need to be this way) "ROM protect" for fast RAM here at least only for the "normal" position ie the second 128K of physical address space of M65.

3. I would really love to see another mapping too, not instead but _also_. That is, the four 64K areas of chip + fast RAM would be mapped for twice for each 64Ks. It can be useful especially when it's needed to "simulate" 64K address space wrapping around behaviour used with linear addressing mode, especially my strange ideas to emulate 8080/Z80 from software, where it's an issue currently too (surely it can be handled by software but _much_ slower then).

4. I'm really not sure how this can be done ... I saw things like "virtual pages" in the IOmap.txt, but I have no idea what it means. Anyway, for many purposes I would also love to use a possibility for having a 64K somewhere in the higher physical memory area (and again, if possible, duplicated, so a next 64K with the same content too, for the reason see my point 3.) Maybe this should be done differently, but my idea, to have four 16K areas allow to see any of the 16K slices of fast+chip RAM there (16K aligned seems to be enough) with also the possibility to make any of the 4*16K mapped as read-only or not at there at least. It would solve many problems, like emulation of other machines but even for porting software and OSes to the M65, as many of them require something like this possibility. It even includes the easier porting of some C64DTV apps (surely it won't happen magically, it's only a minor similarity then!!!!), but even Z80 based GUI-aware, multitasking and micro kernel message passing architecured SymbOS, where M65 is fast enough to actually emulate a Z80 for it, but this Z80 emulation does not allow too much this 4*16K or any similar memory paging it needs. it also true for emulating many other 8 bit machines by software, btw.

gardners, 2017-04-28:
This is quite easy to do. I will simply cause the memory blocks to have a 2nd mapping at some other place in the 28-bit address space.  I will likely provide it as a simple 256KB long linear mapping for chipram + "ROM".

rdpeake, 2021-02-05:
closing this in favour of #301 as that part will be addressed

closed 2021-02-05 by rdpeake