MEGA65/mega65-core #41: no compile - add m65fdisk src-files,

Ben-401, 2017-11-20:


Ben-401, 2017-11-20:
using the m65pcb-ports branch, and at toplevel, "make", getting to the stage where:
- kickup is compiled, 
- on_screen_keyboard_gen etc is compiled,
- BANNER-M65 is compiled/generated,
then when an attempt to compile the "**mega65-fdisk**" PRG, it fails because the source-files are not in the directory.

Unsure where to get these.
The tail of the make-log is below.

```
src/tools/pngprepare/pngprepare logo bin/mega65_320x64_128colour.png sdcard-files/BANNER.M65
argv[0]=src/tools/pngprepare/pngprepare
argv[1]=logo
argv[2]=bin/mega65_320x64_128colour.png
argv[3]=sdcard-files/BANNER.M65
Reading bin/mega65_320x64_128colour.png
Input-file is: width=320, height=64.
Input-file is read and now closed
Processing with mode=0 and output=sdcard-files/BANNER.M65
mode=0 (logo)
Writing out palette of 87 values
done
src/tools/mempacker/mempacker -n shadowram -s 131071 -f src/vhdl/shadowram.vhdl sdcard-files/BANNER.M65@3D00
131071 bytes written
gcc -Wall -g -std=c99 -o src/tools/utilpacker/utilpacker src/tools/utilpacker/utilpacker.c
../Ophis/bin/ophis -4 src/border.a65 -l bin/border.list -m bin/border.map -o bin/border.prg
Assembly complete: 92 bytes output (12 code, 71 data, 9 filler)
( cd src/mega65-fdisk ; make )
make[1]: Entering directory '/home/gerb0002/dev/github/mega65/mega65-core-pauls/src/mega65-fdisk'
make[1]: *** No targets specified and no makefile found.  Stop.
make[1]: Leaving directory '/home/gerb0002/dev/github/mega65/mega65-core-pauls/src/mega65-fdisk'
Makefile:237: recipe for target 'src/mega65-fdisk/m65fdisk.prg' failed
make: *** [src/mega65-fdisk/m65fdisk.prg] Error 2
```

gardners, 2017-11-20:
> src/tools/mempacker/mempacker -n shadowram -s 131071 -f src/vhdl/shadowram.vhdl sdcard-files/BANNER.M65@3D00131071 <13%2010%2071> bytes written

git submodule init
git submodule update

before compiling will check them out.  This is standard for nested git
repositories.

Paul.

Ben-401, 2017-11-20:
```
$ **git submodule init**
Submodule 'src/mega65-fdisk' (git@github.com:MEGA65/mega65-fdisk.git) registered for path 'src/mega65-fdisk'
$ **git submodule update**
Cloning into 'src/mega65-fdisk'...
The authenticity of host 'github.com (192.30.255.112)' can't be established.
RSA key fingerprint is SHA256:nThbg6kXUpJWGl7E1IGOCspRomTxdCARLviKw6E5SY8.
Are you sure you want to continue connecting (yes/no)? yes
Warning: Permanently added 'github.com,192.30.255.112' (RSA) to the list of known hosts.
Permission denied (publickey).
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
fatal: clone of 'git@github.com:MEGA65/mega65-fdisk.git' into submodule path 'src/mega65-fdisk' failed
```
i then added git config user.name and user.email, and tried again
```
$ **git submodule update**
Cloning into 'src/mega65-fdisk'...
Warning: Permanently added the RSA host key for IP address '192.30.255.113' to the list of known hosts.
Permission denied (publickey).
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
fatal: clone of 'git@github.com:MEGA65/mega65-fdisk.git' into submodule path 'src/mega65-fdisk' failed
```
Any ideas?

gardners, 2017-11-20:
Hello,

Probably the issue here is that the git submodule references are by ssh
instead of https.  You can try modifying .gitmodules to use an https URL
copied from the repository "clone or download" option.

Paul.

Ben-401, 2017-11-21:
I could not get it to update the submodule.
So I manually cloned into the ./src dir, ie
```
cd src
git clone https://github.com/MEGA65/mega65-fdisk.git
cd ..
```
and the above worked fine.

below is what i tried when modifying the gitmodules file:
```
$ cat .gitmodules 
[submodule "src/mega65-fdisk"]
	path = src/mega65-fdisk
	url = https://github.com/MEGA65/mega65-fdisk.git
$ git submodule update
<error as in above post>
``` 
I do not know how to fix this, but have provided a work-around.

Ben-401, 2018-02-06:
i have found that you need to "git submodule init" after any change to the .gitmodules file.

further, a clean checkout of the mega65-core repo was necessary to allow the modified ".gitmodules" file to work with "git submodule init" && "git submodule update".

refer to "builder" branch based on px100mhz, dated 6-FEB-2018.

gardners, 2018-02-08:
Assuming fixed by pull request #48.

closed 2018-02-08 by gardners