MEGA65/mega65-core #42: no-compile - cc65 dependancies

Ben-401, 2017-11-21:
after the mega65-fdisk issue, i then get an error during compile of the fdisk programs.
To fix, you need to download "cc65", compile it, and place the binaries in "/usr/local/bin".

I suggest the documentation be updated to reflect this dependency.

I used:
```
$ cd .. (to get out of the $git_root dir)
$ git clone https://github.com/cc65/cc65.git
$ cd cc65
$ make
$ sudo cp bin/* /usr/local/bin
$ cd ..
$ cd mega65-core
```
and then the ```make``` continued to execute the XST commands.

gardners, 2017-11-21:
By all means update the documentation for this, and also explaining the
work around for building the fdisk program.

Paul.

gardners, 2018-02-08:
Has this documentation been fixed?

Ben-401, 2018-02-09:
yes is documented now, at head of px100mhz

closed 2018-02-09 by Ben-401