MEGA65/mega65-core #47: Ethernet frame tag mangling feature wishlist

lgblgblgb, 2018-02-02:
Please consider to adding automatic Ethernet tag removal on RX, so user don't need to deal various network setups where VLANs are configured or other. Without this, every network aware program on M65 using its ethernet capability needs deep knowledge on very different technical details what can be on an average ethernet LAN. Also, to making things symmetrical (stripping tags on RX, adding tags on TX, the user always do not need to mess with those tags neither by inspecting the RX buffer, or on filling the TX buffer), it would be very nice to have automatic tagging on TX as well, however I am not sure what information needs to be set up for M65 and how, that he know what it should add then. As far as I know, there can be odd situations where double tagging is used, though I am not so sure, how generic and realistic scenario it is.

gardners, 2018-02-02:
What is referred to here is 802.11Q VLAN tagging. We should support adding
and removing VLAN tags, and checking if they match (VLAN filter), similar
to how we have optional MAC filtering etc.

TX side we just need to pause TX at byte 12, and add in the 4 bytes
(calculating CRC as we go).

On RX side, similarly, pull the 4 bytes out, and reject nonmatching
packets. If a VLAN tag is removed, we can just rewind the write address in
the frame buffer by 4 bytes.

lydon42, 2024-08-17:
No, I don't think the MEGA65 needs to be able to do 802.1Q (I assume 802.11Q @gardners wrote is an typo... 802.11 is WLAN, 802.1Q is VLAN), it is a pure client, the switch before it should handle the VLANs. No need that we put that complexity into the MEGA65 core.

If we want to revisit this, please first discuss on steering-topics.

closed 2024-08-17 by lydon42