MEGA65/mega65-core #49: monitor_load not releasing the serial port

Ben-401, 2018-02-08:
unsure if this is a problem others are facing, but,..

after running:
sudo src/tools/monitor_load -l /dev/ttyUSB1 -b bin/nexys4ddr.bit -k bin/KICKUP.M65 -R MEGA65.ROM -C CHARROM.M65 -4 src/utilities/test01prg.prg 

the serial port seem unusable.
refer to post:
https://groups.google.com/d/msg/c65gs-development/Pmz0s4NfCdk/p7gI-UyGBAAJ

Upon closing the program, should the serial port be reverted back to its previous state as when the monitor_load program started?

gardners, 2018-02-08:
The messages you are seeing suggest that the Linux Kernel is physically detaching the underlying USB connection. This should not be possible for monitor_load (or any other serial port using program) to do, as it is a lower-level function. I do see many problems with the FTDI USB serial drivers, though, causing all sorts of very annoying problems. That, I suspect is where the problem actually lies.
There is in the meantime nothing that I am aware we can do to fix this.

Ben-401, 2018-02-09:
this is not a showstopper, but i do note that the serial port is not closed nicely when the program exits.

closed 2018-02-09 by Ben-401