MEGA65/mega65-core #53: load project in ISE, missing files etc

Ben-401, 2018-02-13:


Ben-401, 2018-02-13:
it seems there are only two ISE-PROJECT files (*.xise), namely "nexys4" and "nexys4ddr".

loading one (or the other) in ISE does not load simply, and results in many conflicting file paths.
this suggests that we are unable to build using the ISE-GUI

Ben-401, 2018-02-13:
for example, the toplevel component in xise is called "container.vhdl" whereas the prj files referrs to it as "nexys4ddr.vhdl"

gardners, 2018-02-13:
I haven't used the ISE GUI for some time. The current build system is based
solely around command line building under ISE.
I don't mind if you want to ressurect GUI building, but the command line
process must remain the authoritative method.

Paul.

rdpeake, 2021-02-05:
Have moved to vivado

closed 2021-02-05 by rdpeake