MEGA65/mega65-core #56: make Makefile compile-script: adding user.vhdl to design

Ben-401, 2018-03-09:
difficulty in knowing how to add a component to the design

using the ISEv14.7 as a glorified text-editor, I have changed iomapper to include a new-component that I built myself. After saving the project etc, the new-component appears under/within the iomapper.
Yes ISE can build the design to include my new-component.
BUT, when building the design using "make bin/lcd4ddr.bit" the build process fails because it cannot find the vhdl-file for the new-component.

it seems i need to go into the Makefile and specify the path to the  vhdl-file of the new-component.

- the previous build system (using compile.sh) made use of the ISE-project-file to know what source files are required to synthesise the design.
- the new Makefile process requires you to not only add your component (and modify parent-component.vhdl) in ISE, but also in the Makefile somewhere.

Currently I am unsure where to modify the Makefile to include my new-component.

Yes I understand that PGS wants the compile/build to be performed via the command line, specifically using his Makefile (and run_ise), and this is the authoritive method of compilation, BUT that system does not easilly allow a new user to come along and add their new-component to the design, without having knowledge of the build system.
Most/all new fpga/vhdl people to the team will be familiar with ISE, so I suggest that the Makefile/run_ise process looks to the ISE-project-file for what files to load to be able to synthesize.

Just an idea, what do you think?

gardners, 2018-03-09:
Hello,

The Makefile should be considered to be the authoritative source. To add a
file, add it to one of the lists of VHDL files in the Makefile, and then to
the various .prj files in isework/.  You should then be able to build with
make.  If you don't add it to the Makefile, the main symptom will be that
simulation won't work from the makefile, and that if you change the source
file and rerun make, it won't know that anything has changed.

Generally, I would recommend against using the ISE editor because of the
above.

Paul.

rdpeake, 2021-02-05:
closing as using vivado now over ISE

closed 2021-02-05 by rdpeake