MEGA65/mega65-core #72: RESET approach in the Mega65 - suggestion

adtbm, 2019-03-19:
Testing the latest Bitstream and the soft Reset implementation
i thought about the main approach of doing a Reset on the Mega65.

How is it about, of making it possible, to swap inside the Freeze menu, between a soft reset and a Reset, without module check.
So a kind of a Hard Reset, maybe toggle between the two by pressing "R" and swap the "F5" key description accordingly between (F5 - (S)-Reset) and (F5 - (H)-Reset)

This would result, that the user would have the possibility of 3 different Reset functions , excluding Powering Off/On the Mega65:

1 x Soft reset (like shortening pin 1 and 3 on the c64 userport or typing SYS64738) 
      - via Freeze Menu (S)-Reset
1 x Hard Reset (excluding the module check) 
      - via Freeze Menu (H)-Reset
1 x ultra Hard Reset/Reboot 
      - via Reset button on the Mega65

adtbm, 2019-03-19:
Issue is solved after speaking with Paul.

planned update:
Whenever doing a Reset (pressing " F5" inside the freeze menu), it will be checked for a module signature at $8000-$8009. If one exists, then the user will be asked, if it should be ignored or not and the Reset will be done accordingly.
"F6"-Key could be used for Hard Reset.

rdpeake, 2021-02-05:
Moved to mega-freezemenu as freezer now lives over there

closed 2021-02-05 by rdpeake