MEGA65/mega65-core #78: FREEZER.M65 not built in "all" target

vvrbanc, 2019-04-29:
This file needs to be on the sdcard in order for the freezer to work, but it is not built at all.

Ideally it should also be built into the bitstream, and monitor_load should be able to sideload it too. Or maybe embed it into the hickup.m65?

gardners, 2019-04-29:
It can't live in the bistream, as there is no free BRAM -- we are using all BRAM to make the memory for the machine.  Similarly, HICKUP.M65 is just a replacement for the HYPPO Hypervisor memory, which is only 16KB in size.

It can already be side-loaded onto an SD card using mega65_ftp.

Basically the freezer requires a working SD card to function, and there is no way to work around that right now.  Later we may add support for freezing to expansion RAM, once we have the R2 PCBs and the Hyperram integrated.

But in the short-term, we can add a rule to make the FREEZER.M65 binary to put in the sdcard-files directory.

Longer-term, we can modify MEGA65 FDISK to automatically install a version of it when formatting the SD card, in the same way that it does now with the ROM file, if a ROM is already loaded into memory when it runs.

gardners, 2019-04-29:
Build target added in commit 38baadef

gardners, 2019-04-29:
See commit above. Last remaining step is to modify FDISK to automatically find the freezer in memory and write it out to the SD card when formatting.

gardners, 2019-04-29:
Ok, so in theory I have implemented this, but not yet tested any of it.  
Can you do a synthesis run, and try it out and let me know how it goes for you?

vvrbanc, 2019-04-29:
For sure!

vvrbanc, 2019-04-29:
Build fails with "no rule to make target freezer.s".

I believe it is because freezer.h is missing. File is not in repo but referenced in freezer.c

gardners, 2019-04-29:
Thanks! (And fixed)

vvrbanc, 2019-04-29:
FREEZER.M65 builds successfully from main Makefile now (as of commit dbf5a26a52ca).
I think we can close this one and reconsider what to do with it once the R2 PCBs are out.

gardners, 2019-04-29:
Did you try running with the bitstream, running FDISK and seeing if it writes the FREEZER.M65 file to the SD card automatically?

vvrbanc, 2019-04-29:
Tried just now, got corruption on SD card. As in, directory entries garbled. I'll test whether it's because of that preload thing you added or something I did with the renames maybe.

I'll get back to you once I find out which commit broke it.

gardners, 2019-04-29:
Thanks. Let me know what you discover. It's possible that unrelated changes to FDISK I made over the weekend to get it compiling natively under linux have broken things.

vvrbanc, 2019-04-29:
Tried commenting out $(SDCARD_DIR)/FREEZER.M65@12000 from the Makefile in 52e28c1 but still got corruption.

Commit 3f836c8a works, so it's either the fdisk changes or the renaming. I'll investigate further and post my findings.

gardners, 2019-04-29:
I suspect it is most likely the fdisk changes.

Paul.

vvrbanc, 2019-04-29:
Looks like it's the fdisk. No corruption with mega65-core/development HEAD and older mega65-fdisk 14d6fdac53

There is a slight difference in fdisk output on the newer version, and also screen border goes red when operation completes. I guess that means error? No error message. Anything useful I can get out of monitor?

Old version says:
Writing 34    nt loaded rom... 
First free cluster is 3

New version says:
Writing 31   nt loaded rom...
First free cluster is 128

The reset of the output is identical.

gardners, 2019-04-29:
The change in first free cluster is expected, because I changed the allocation routine to search for a sector with no allocated clusters.

I'll have to test it here when I get a chance. Alternatively, if you are able to do both runs and make a compressed image of the disk image on both occassions, it will be easier for me to work out the cause of the corruption, e.g., if a FAT sector or part of the ROM is being written over the directory.

vvrbanc, 2019-04-30:
I made 3 images:

https://vrbanc.com/old-clean.img.bz2
https://vrbanc.com/new-clean.img.bz2
https://vrbanc.com/new-corrupted.img.bz2

Compressed images are around 1MB each. Uncompressed size is 16GB, I.E a full dd dump of the sdcard.

New images are made with:
mega65-core/development 52e28c1fa0
mega65-fdisk/master f4c211cd9c

Old image is made with:
mega65-core/develpment 52e28c1fa0 (same as the new ones)
mega65-fdisk/master 14d6fdac53 (last commit prior to Apr27)

In all cases I pre-loaded the ROM with monitor_load, so it was written out to the root directory by fdisk.

Prior to creating old/new-clean.img I zeroed out the sdcard with dd.

The new-corrupted.img is made by first formatting it with old commit, then formatting again with the new commit. It shows the mega65.rom file and then a bunch of garbage in the root directory.

The new-clean.img doesn't look corrupted when inspected with mega65-ftp, however it takes unusually long to get the directory listing (3-4 seconds).

Additionally, fsck reports errors with the new-clean.img. 

When comparing the hexdumps of new-clean.img and new-corrupted.img, they are identical, except the corrupted one has some non-zero data at hex offset 1af6600-1b16400, which seems to be old data parsed as a directory entry.

Comparing hexdump of new-clean.img and old-clean.img show some odd stuff on the new one, but I don't know enough about fat32 to be able to tell what exactly is wrong.

So, in summary... 

- new fdisk creates a filesystem with errors, even on a zeroed out sdcard
- new fdisk doesn't properly clean existing data, or (more likely) has bad data pointers. Somehow it shows data from previous format as directory entries.

gardners, 2019-05-01:
Thanks. I agree the problem is almost certainly wrong data pointers. I'll have a look as soon as I get a chance.  Hopefully it will be some stupid simple error.

gardners, 2019-05-01:
ok, the slow directory reading is a bit of a clue: It is the FAT sectors that are getting corrupted.

gardners, 2019-05-01:
Ok found obvious cause of problem. Just pushed a commit now. Please retest. Hopefully will all be good.

vvrbanc, 2019-05-02:
Looks better, but not quite there. There is an error in the procedure that writes out the ROM file.

Clean format (without ROM loaded) works fine, sdcard is properly formatted, chkdsk/fsck finds no errors.

However, if a ROM is loaded and written out, it does something bad. The sdcard has filesystem errors, see here: http://dpaste.com/20SYBZ7.txt

Again I made 2 images, one created with old fdisk, one with new, rom written out in both cases.

Hex dump diff screencap here: https://vrbanc.com/imgdiff.png

Left is old, right is new.

In the left one, first cluster for MEGA65.ROM is 03, then chained up from 4 to 22. Seems ok. 

In the right one, first cluster for MEGA65.ROM is 80. Offset seems correct, file actually starts at 0x1b73400. (0x1b73400-0x1af5400)/4096 = 7F.

But the new chains look wrong in 2 ways:
1) they are offset by 0x200 from beginning of FAT.
2) the chain starts at 01. Shouldn't it start at 81?

Also, in the new image there seems to be some garbage at 0x1af7400 and 0x01af8400 that doesn't seem to be present in the old image.

gardners, 2019-05-02:
First, thanks for your thorough investigation into this.
You are right that the cluster numbers are wrong. It should actually begin at cluster $80.  Let me have a quick poke at it.

gardners, 2019-05-02:
Ok. Found and fixed.  Now I get:

```
sudo fsck.vfat -n /dev/loop1p1
fsck.fat 4.1 (2017-01-24)
Warning: Filesystem is FAT32 according to fat_length and fat32_length fields,
  but has only 8105 clusters, less than the required minimum of 65525.
  This may lead to problems on some systems.
Free cluster summary wrong (8104 vs. really 8072)
  Auto-correcting.
Leaving filesystem unchanged.
/dev/loop1p1: 2 files, 33/8105 clusters
```

and 

```sudo mount /dev/loop1p1 /mnt
paul@F96NG92-L:~/Projects/mega65/mega65-core/src/mega65-fdisk$ ls /mnt
MEGA65.ROM
$ diff MEGA65.ROM /mnt/MEGA65.ROM 
```

gardners, 2019-05-02:
Please let me know if that fixes it for you, too.

vvrbanc, 2019-05-05:
The offset numbers are now fine, but I'm still seeing errors. Now the problem is this:

00d1e400  81 00 81 00 82 00 82 00  83 00 83 00 84 00 84 00

This should be
00d1e400  81 00 00 00 82 00 00 00  83 00 00 00 84 00 00 00

I believe the error is here
https://github.com/MEGA65/mega65-fdisk/blob/4f5fb3d13ef74d48ba968223020f596e5be87109/fdisk_fat32.c#L73

This should be "unsigned long" (32 bits) and not "int" (16 bits).

I tested this and it seems to fix the file errors.

fsck still complains about the number of free clusters, I'll try and get to the bottom of that too.

gardners, 2019-05-05:
Thanks for that. The wrong number of free clusters would be because the MEGA65 simply ignores this value at the moment. But of course it wouldn't be hard to read the sector in question, subtract the number of allocated clusters and write it back.

vvrbanc, 2019-05-05:
looks good now, I think we can close this issue

closed 2019-05-05 by vvrbanc