MEGA65/mega65-core #85: Makefile needs to account for multiple Vivado locations

Jordanis99, 2019-05-16:
To reduce issues during the compilation of the bin/te0725.bit target, the Makefile script could be edited to check whether Vivado exists in either the /opt/Xilinx folder (which the script currently assumes Vivado to be in), or the /tools/Xilinx folder, and adjust the following commands in the script accordingly. 
This is preferable because in older versions of Vivado, Vivado automatically installed in the /opt/Xilinx folder, however in the current version of Vivado, the Linux web installer defaults the installation location to the /tools/Xilinx folder. Therefore this Makefile edit would be beneficial to support both older and current versions of Vivado to prevent installation to an unusable location; which could lead to re-installation, symbolic linking, or manual editing of the Makefile on an individual basis.

thomotron, 2019-05-17:
I suggest that the `vivado_wrapper.sh` script check if an environment variable with the Vivado path is already defined and use that, falling back to the default if necessary. `make` can override environment variables if they're [included in the command](https://stackoverflow.com/a/3169141) so perhaps that might be useful?

gardners, 2019-05-22:
A fine suggestion. would you like to propose a patch to implement this?

closed 2021-02-05 by rdpeake