MEGA65/mega65-core #86: Vivado Installation on Ubuntu 19.04

SirOggy, 2019-05-17:
Hello everyone,
Although not an issue with the Mega65 project itself, I have encountered an issue while trying to install Vivado 18.03 on Ubuntu 19.04 Disco Dingo.
It seems that Vivado hangings indefinitely on the 'Generating Device List...', refusing to install correctly.

If one cancels the installation, sometime the installed files will be present, however inoperable.

If one tried to copy across Vivado files from another Ubuntu system running 18.04 to another running 19.04 and performing the installation, which while faster initially, will once again be snagged by the 'Generating Device List...' issue.

I have reproduced the issue multiple times on my local machine locally and on virtual machines running 19.04.

Some help was provided online by searching for the hanging issue, however with no avail.
[An example of the issue occur in earlier versions of Vivado](https://forums.xilinx.com/t5/Installation-and-Licensing/Intaller-hans-at-quot-generating-installed-device-list-quot-in/td-p/658743)
It's only when one downgrades to Ubuntu 18.04 or 18.10 that the installation succeeds.

Therefore, I believe it should be considered that until Vivado officially supports new versions of Ubuntu, anything later than 18.10, that the Mega65 project advise people to seer clear of anything newer than 18.10.

If anyone has critique or suggestions, I'm open to them.
Thank you.

gardners, 2019-05-22:
Thanks for exploring this. Perhaps propose a pull request to the build instructions indicating that this is the case, and refernce this issue, so that if it gets fixed later, people can find the info.

Paul.

rdpeake, 2021-02-05:
Vivado is installable on 20.04 (by paul today)

closed 2021-02-05 by rdpeake