MEGA65/mega65-core #89: Cant compile from make file

piek0012, 2019-05-17:
yer cant compile file need help
Got some errors saying need some target file from make file 
not making sense
fixed all theMissing packages, etc....

piek0012, 2019-05-17:
I think the make file is a issue lol

piek0012, 2019-05-17:
Comrades cant fix the crap either lol

piek0012, 2019-05-17:
fixing the missing packages is already enough for me

vvrbanc, 2019-05-17:
Hi, are you using the "development" branch? If so, what error are you getting?

piek0012, 2019-05-20:
Sorry for the late response, I believe I am using the development branch. After installing the missing packages, I am getting errors from the makefile, I did modify some parts before. I'm not sure if im progessing or have made things worse.

My errors are: 

Makefile:559: warning: overriding commands for target `bin/COLOURRAM.BIN'
Makefile:553: warning: ignoring old commands for target `bin/COLOURRAM.BIN'
Makefile:629: warning: overriding commands for target `src/tools/monitor_load'
Makefile:611: warning: ignoring old commands for target `src/tools/monitor_load'
make: *** No rule to make target `src/tools/mega65-fdisk/m65disk.prg', needed by `bin/COLOURRAM.BIN'. Stop

piek0012, 2019-06-06:
The solution was to declare the integer outside the function in the make files for fdisk.

gardners, 2019-06-06:
Please indicate the specific change that is required, preferably with a pull request.

rdpeake, 2021-02-05:
Currently building without issues, reopen if this still persists, or start new issue if there are new issues.

closed 2021-02-05 by rdpeake