MEGA65/mega65-core #96: ADC and SBC 32 bit do not take note of Carry/Borrow in

paulscottrobson, 2019-08-02:


paulscottrobson, 2019-08-02:
Lines 5709/5724 of gs4510.vhd ; calculation does not take account of current state of carry flag.

lgblgblgb, 2019-09-04:
That's interesting. Historically 65xx CPUs does not have ADD but ADC only, and no SUB but SBC only, so you must set/clear carry first if no need for carry, etc, as we all know. But for 32 bit thing, I think, it's actually quite rare for the need to have longer data type, ie 64 bit ADD/SBC where it's important to take account the carry from the lower DWORD, so probably it's often just 32 bit, thus also needing to manage carry flag before is not so much important and rare to be used, and maybe thus more efficient to assume no carry is taken account on 32 bit ADC/SBC as this is the most likely scenario what this will be used for. But this is only my opinion, for sure ... So maybe this is not even a bug but "the right thing to do" ... ? Though it must be called ADD and SUB 32 bit then in asm syntax to avoid confusion, if we ever have an assembler at least knowing these M65 specific opcodes ....

paulscottrobson, 2019-09-04:
I think it's logically consistent.

gardners, 2019-09-04:
Fixed by making ADC32 and SBC32 honour the C flag.

closed 2019-09-04 by gardners