MEGA65/mega65-core #97: Include FREEZER.M65 as memory content in bitsream + make FDISK to write it out

lgblgblgb, 2019-08-11:
Minor enhancement idea (as discussed with Paul): please include FREEZER.M65 memory content somewhere in the bitstream, and also include code in FDISK so it can write it out. Thus, the user does not need to deal with manually placing that file as a post-preparation work, also it reduces the possible problems to have incompatible FREEZER.M65 put on the card by mistake, which does not work with the rest of the system too well.

rdpeake, 2021-02-05:
Implemented

rdpeake, 2021-02-07:
Fdisk tool does not write out freezer.m65 as It fails to correctly identify it in memory due to looking for the wrong start sequence.

gardners, 2021-02-07:
ah, ok. possibly if we have switched to having it compressed with exomiser.

Paul.

lydon42, 2022-01-04:
Essentially this is what the new COR + FILES combination is, so it could be closed?

closed 2024-08-17 by lydon42