MEGA65/mega65-core #99: Easyflash & Easyflash3 not programmable from MEGA

adtbm, 2019-09-11:
The easyflash seems to work but if you try to flash a .CRT via Easyprogrammer.
It gives the Message: "A flash chip does not work or is not supported by the EasyAPI flash driver."

adtbm, 2021-02-05:
This is still valid.
I think this individual bug belongs to the whole Cartridge story. That cartridges are not working, are not supported, etc.
At the moment the MEGA65 cartridge port is in a state, that Eazyflash is not working anymore at all.

lydon42, 2022-05-29:
This is a #564

lydon42, 2023-04-10:
sy2002 managed to get the programming of EF 1CR going using the C64 core. This was all a big bunch of timing problems on the cartridge bus. So probably we can migrate this to the mega core.

lydon42, 2024-08-17:
Update check: this is still true, EasyFlash/EasyProg does not recognize the EF1CR from C64 mode. C64 Core has no problem and can program the cartridge just fine, so that is the current workaround.

lydon42, 2024-08-18:
I am closing this in favor of #692 as they describe the same problem, but the new issue has some useful information inside, while this one got lost in time.

closed 2024-08-18 by lydon42