MEGA65/mega65-core #100: SDHC microSD issues

adtbm, 2019-09-12:
- When i fill the SDcard with tons of images (more than 100), in the discchoser (inside freeze menu) at a certain point the disc choser only shows "@@@@@@@@@@@@@@@@@" instead of disc image name.

- when i copy and delete several images to the SDcard and then add some later, the images are not in alphabetical order anymore. Sometimes, they do not show up in disc choser at all anymore. Or the image is corrupted and i can't load the files or mount the image.
Only way to get around this is to make a backup of the SDcard to your PC then reformat the whole SDcard in MEGA65 SD format Utility, then copy all files back. Then the images are working again and show up in alphabetical order, but the number of images visible is still limited.

rdpeake, 2021-02-05:
Can this be checked if still an issue and if so, can steps to reproduce be supplied

adtbm, 2021-02-05:
The first part seems to be solved.

the second part is still valid.
iirc Paul said this is a problem with the FAT.
It is easy reproducable, if you copy a .D81 image to the SDcard, it will show up in the "disc-chooser" in the FREEZE menu
at the bottom and will not be sorted to it's alphabetical position.
Also the rest is still valid. sometimes when a File on the SD has not been properly updated in the FAT, the D81 shows in the disc chooser, but it's not possible to show or load the content.

The work around above is still valid and working.

lydon42, 2024-09-10:
Yes freezer does not sort filenames, but this is no core issue, it rather should be a freezer issue. Opened one for you,  closing this.

closed 2024-09-10 by lydon42