MEGA65/mega65-core #104: CIA timer problem

lgblgblgb, 2019-09-19:
During my experiment with IP65 TCP/IP stack I found a problem. IP65 needs some timing info, and it uses CIA timers for that purpose. It seems it fails on M65 but works in C64 (at least tested with VICE by me). The actual code what IP65 uses can be seen here:

https://raw.githubusercontent.com/cc65/ip65/master/drivers/c64timer.s

I quickly hacked some lame test program to isolate these functions only.
[c64timer_test.zip](https://github.com/MEGA65/mega65-core/files/3633031/c64timer_test.zip)

It simply shows the values given by those functions. On C64 the first 16 bit number normally increases, on M65 it's always zero. The second single byte is the second counter from TOD, that seems work on both of C64 and M65.

gardners, 2019-09-20:
The problem seems to be related to timer b counting timer a underflows.  The logic in the VHDL was crazy, and I have simplified it.

lgblgblgb, 2019-09-20:
As far as I see it fixed the problem by looking the result of my test program and compared with C64 (using VICE). The ethernet thingy we talk about is another question, not related this one, I guess :) Thanks.

lgblgblgb, 2019-09-21:
Still with CIA stuff, I measured 20sec of TOD second counting on the screen as about 29sec of real time with the stop watch app of my smart phone :) So it seems M65 TOD update is about 50% slower than it should be? But also NTSC/PAL stuff can effect this maybe (I am not sure) bit not as much as here I noticed, probably. Interestingly the timer value looks like in pair what VICE/x64 shows just the TOD looks slower.

lgblgblgb, 2019-09-21:
Interestingly even on VICE 30 sec real of "TOD time" is 36 sec real time, though I tried with PAL/NTSC, sounds interesting (real CIA - AFAIK - would maintain TOD clock from the 50/60Hz sampled from the power supply 9V AC, IIRC so it depends on the region of frequency of the mains ...). But with M65 the difference looks bigger anyway.

closed 2021-02-28 by rdpeake