MEGA65/mega65-core #106: CIA TOD seems to be too slow

lgblgblgb, 2019-09-21:
Still with CIA stuff, I measured 20sec of TOD second counting on the screen as about 29sec of real time with the stop watch app of my smart phone :) So it seems M65 TOD update is about 50% slower than it should be? But also NTSC/PAL stuff can effect this maybe (I am not sure) bit not as much as here I noticed, probably. Interestingly the timer value looks like in pair what VICE/x64 shows just the TOD looks slower.

Interestingly even on VICE 30 sec real of "TOD time" is 36 sec real time, though I tried with PAL/NTSC, sounds interesting (real CIA - AFAIK - would maintain TOD clock from the 50/60Hz sampled from the power supply 9V AC, IIRC so it depends on the region of frequency of the mains ...). But with M65 the difference looks bigger anyway.

gardners, 2020-02-29:
Confirmed that problem is real.

gardners, 2020-02-29:
Above commit patches i2cstatus.c to also show CIA TOD with RTC, so that we can very easily compare speeds.

Ideally we should switch between 50Hz and 60Hz clock, based on PAL vs NTSC.

gardners, 2020-02-29:
Above commits should fix the remaining known discrepencies.

frehwagen, 2020-03-16:
Fixed and successfully tested together with #163, merged back to stable_base so this is going to close now.

closed 2020-03-16 by frehwagen