MEGA65/mega65-core #108: CharROM size is too small

andrewoesten, 2019-09-22:
As said in my E-Mail, the CharROM size in the memory map seems to be too small.

lgblgblgb, 2019-09-23:
Aaaaa, now I understand why you extended the char "WOM" size in your Xemu fork ;)

lydon42, 2024-08-17:
FREEZER can load alternate charsets, BASIC65 can redefine characters. Just load 4k of the charset. In addition this sounds partially like it was an xemu issue?

Closing because it is unclear what is meant. Please create a new issue with better examples and explanation (not referring to an email I can't see).

gardners, 2024-08-17:
I believe this is the CROM bit referred to in #309. Confirming that we are not fixing it, as there are good alternatives, and it would cost a BRAM, that we don't have spare on A100T targets