MEGA65/mega65-core #114: PAL/NTSC Cycle-Accurate and HDMI-Compatible video modes needed

gardners, 2019-11-03:
The current video modes are VGA-oriented, and have  a number of problems:
1. They are not 63/65 VIC-II cycles per raster-line.
2. They have the wrong number of raster lines.
3. They use different dot clocks from one another, which causes shimmering in different modes, because the buffering for this is rather borked.
4. They won't work on HDMI, for many of the above reasons.

Thus we need to switch to the 27MHz dotclock HDTV modes for PAL and NTSC, but ideally in a way that still works on most VGA monitors. It may not be possible to make it work on all VGA monitors.

VGA output will likely need to be masked to 720/640 pixels wide, to avoid wrong colour correction.

There is significant work on this already on the unstable branch.

gardners, 2019-11-04:
The main remaining problem is that the video image appears on the wrong place on many monitors. This likely comes from the fact that we are producing a 800 pixel wide image, but the VGA modes want only 720.  So we should make sure that those front and back porches of 40 pixels either side are black.  But we want at the same time to provide the full 800 pixels wide for the MEGAphone. Thus the commits above provide two output streams, one 800 wide and the other narrower, as required.

gardners, 2019-11-06:
My LG TV thinks the 720x480 60Hz NTSC mode via VGA is 640x350 60Hz, with screen shifted down and right a bit, as well of course of looking way too large, so only part of the screen fits on.
Same LG TV refuses to even display the 720x576 50Hz PAL mode.

gardners, 2019-11-06:
Confirmed HSYNC and VSYNC polarity has no effect on these problems.

gardners, 2019-11-09:
The work on the unstable branch to get HDMI output working is progressing.
It is now at the point where running the src/utilities/hdmitest.a65 program and pressing SPACE a couple of times will configure the ADV7511
correctly, including retreive EDID info. The effect is that a connected HDMI monitor will begin receiving an HDMI signal.  The problem is that the display, however, remains blank.

gardners, 2019-11-13:
Considerable further work has been done on the unstable branch, such that we now have working HDMI output, and also VGA output that seems to work on a majority of monitors.
Deft etc have undertaken initial testing of this, and it seems to be functional, and thus can be merged into the 114 feature branch, in preparation for closing this issue.

gardners, 2019-11-16:
A sub-problem is that the $D012 raster counter is still not counting in the VSYNC period, which breaks some things. This is being fixed also.

gardners, 2019-11-27:
See also #146

adtbm, 2019-12-07:
Tested Bitstream on Nexys4 & Widget.

- 1st monitor: (LCD Panel) Video looks good:
PAL has a black frame on the right and upper side:

[image]

NTSC looks OK:
[image]

Older VGA monitors didn't received a signal at all.

adtbm, 2019-12-07:
Test with R2 board, 
bitstream: mega65r2-20191202.19-HEAD_n-b03d929+DIRTY.bit

Attached R2 to HD 720p Beamer.
But when pressing the RESET Button on the MEGA65 reboots and shows for a splitsecond , the MEGA65 display Look Here:

frehwagen, 2019-12-21:
Finally this has been merged to stable_base branch. All open finding has been separated into new issues, so lets close this.

closed 2019-12-21 by frehwagen