MEGA65/mega65-core #115: Fix keyboard rework in unstable branch

gardners, 2019-11-03:
The keyboard matrix stuff in keyboard_complex was switched from using a 72-bit parallel interface to a little RAM block. This saves LOTS of FPGA resources. Unfortunately, right now it causes problems with the Widget board.  The exact problem needs to be identified and then dealt with.

gardners, 2019-11-06:
The problem is bad keyboard behaviour with widget board, and with any target when CPU is at 40MHz.
To reproduce, try typing using widget board, and it will sometimes print the wrong characters.
Also, if you POKE0,65 to set the CPU to 40MHz, the keyboard on any target will be very horrible to use.

gardners, 2019-11-06:
Confirmed resolved for nexys4ddr-widget target

gardners, 2019-11-06:
Confirmed fixed on mega65r2 target with real keyboard.

I now consider this problem fixed.

adtbm, 2019-11-08:
OK i can confirm #115 with freshly formated SD card on NEXYS4DDR+WIDGET
- naming internal incorrect
    name of bitstream on Scryptos: nexys4ddr-widget-20191105.02-HEAD_n-13bf229+DIRTY.bit
    name internally during bootup: (.13BF229+DIRTY.20191104.23
- c64/c65 keyboard works in 40 MHz
- Freezer menu crashes on leaving (entering works)
- loading d81 works
- NTSC video looks good
- PAL can't test (Freezer crashes)

frehwagen, 2019-11-08:
Fixed on stable_base build #2 by merge commit https://github.com/MEGA65/mega65-core/commit/98f8500eb2f50836b2baaf0fb45feef18f871c09, finally we are able to close this issue.

closed 2019-11-08 by frehwagen