MEGA65/mega65-core #116: Fix D81 / DMD disk image handling in unstable branch

gardners, 2019-11-03:
D81 mounting is currently not working on the unstable branch.
Find problem and fix.
Suspect the recently implemented MEGAdisk support as the cause.

frehwagen, 2019-11-11:
While it seems that the D81 mount issue was a broken image file on sd card, we will use this ticket to properly integrate MEGAdisk support into stable-base branch.

gardners, 2019-11-13:
Fixed by commit ba65cb53efdcb3535bdad927a2147541cc002a86

closed 2019-11-13 by gardners

gardners, 2019-11-27:
Proposed test procedure:
1. Use freezer to set drive 0 and 1 both to "- no disk -".
2. DIR U8 -> should give ?DOS: DRIVE NOT READY
3. DIR U9 -> should give ?DOS: DRIVE NOT READY
4. Use freezer to set drive 0 to "- internal 3.5" -"
5. DIR U9 -> should give ?DOS: DRIVE NOT READY
6. DIR U8 -> should show directory of disk in drive, or hang if no disk inserted (See #147 )
7. Reset (solves hang problem if you had no disk to insert in (6)).
8. Use freezer to set drive 0 to a disk image, and drive 1 to "- no disk -"
9. DIR U8 -> should show directory of selected disk image
10. DIR U9 -> should give ?DOS: DRIVE NOT READY
11. Use freezer to set drive 1 to a disk image, and drive 0 to "- no disk -"
12. DIR U8 -> should give ?DOS: DRIVE NOT READY
13. DIR U9 -> should show directory of selected disk image
14. Use freezer to set drive 0 to one disk image, and drive 1 to another disk image.
15. DIR U8 -> should show directory of first disk image selected
16. DIR U9 -> should show directory of second disk image selected