MEGA65/mega65-core #118: Check joystick pot input circuitry on MEGA65r2 PCB for completeness and correctness

gardners, 2019-11-03:
We should ;probably try to do this sooner rather than later, so that any fix can be incorporated into the r3 PCB for the devkits.

adtbm, 2019-11-04:
#122

gardners, 2019-11-04:
It needs to work with 1351 mouses and paddles.  The mouses in particular require behaviour that matches the C64, i.e., using a threshold comparitor and having a drain driver for pulling the line low.

gardners, 2020-02-13:
It looks like all we need to add is some 1.2nF capacitors between the POT lines and GND on the joystick ports.

gardners, 2020-02-13:
Looking more closely at the JA_POTX (and friends) lines on the FPGA, it looks like by substituting R77 for 0 Ohm, C130 for 1.2nF, removing R75 and R73 would get the circuit mostly right, except that it would not have any voltage conversion from 5V down to 3.3V.

gardners, 2020-02-13:
Ideally the circuit would be quite different, however:  

1. JA_AX would be connected to the top of a transistor that when enabled, would pull the line to ground.  The enable line for this transistor would be connected to some pin on the FPGA.  This pin can be shared for all four pot lines.
2. JA_AX would also be connected to a 1.2nF capacitor whose other pin is connected to GND.
3. JA_AX would also be connected to the input of a 5.5V to 3.3V level converter (or something else that can limit the voltage to no more than 3.3V.

For testing, this could be perhaps be bodged by removing R73, R77, R75 and C130, and then using those pads to connect the necessary circuitry to.
Alternatively, we could make an adapter for the joystick ports that contains the necessary circuitry inline.

Either way, a first step could be to implement the circuit on a PMOD board.

gardners, 2020-02-14:
Found this specifications document for the 5717 chip in the 1351 mouse:

http://cbm-hackers.2304266.n4.nabble.com/attachment/4660682/0/5717spec.txt

Also confirmed with gnucap that the current configuration basically varies between 0V and about 0.85V when fed with between 0V and 5V as input.  So safe to use as is, just won't work with the 1351 mouse.

From the 5717 documentation: The SYNC input (which is just connected to a POT line with a 5.1K pull-up, as far as I can tell) must be <=1V to be considered low, and >=4V to be considered high.

I can see no simple way to replicate this by just changing values of the existing passives on the POT lines, that doesn't also result in >3.3V on the FPGA pins.

In short, we need to have the level converter and transistor method.

gardners, 2020-02-14:
This means adding a NC7SZ126P5X and transistor for each of the four pot lines, removing the resistors and capacitors currently there, and having. As there are already enough signals routed to the joystick ports, this should hopefully not require significant re-routing of the PCB.

The challenge is how to test before the board gets re-spun, so that we can be 100% sure it will work.

The various parts of this have been done before in:

http://c65gs.blogspot.com/2018/01/working-on-paddles-and-mouse-support.html

We should indeed make up a PMOD that implements an entire joystick port, and test it out.

closed 2021-02-09 by rdpeake