MEGA65/mega65-core #126: Freeze menu should offer disk chooser for both drives

gardners, 2019-11-06:
Currently you can't choose a disk image for the 2nd internal drive.  This should be fixed.

gardners, 2019-11-15:
Fixed in unstable branch + updates to freeze menu.

bluewaysw, 2019-11-19:
This is beeing handled on the 116- branch together with issue #116 related changes.
Just decided to include 2 real floppy support to the branch as well, because it already contains a lot of fixes that relate to the dual drive support in freezer/disk chooser.

adtbm, 2019-11-25:
reviewing #116 issue. 
i have encountered issues mounting images.
in my example i mounted an image to drive ID#0 and an Image to Drive ID#2
it resulted, that the internal 3.5 floppy drive was accessed while loading "$",9 NOT the image i've mounted

1st picture shows current mounting state after powering on:
[image]

2nd picture shows image loaded for ID 0
[image]

3rd picture shows image loaded for ID 1
[image]

4th picture shows mounting state after mounting images to 0 & 1 -looks ok-
[image]

5th picture shows correct mounted image after LOAD"$",8   -OK-
[image]

6th picture shows INCORRECT mounted drive after LOAD"$",9   -NOT OK-
It shows the content of the 3.5 floppy disc.

so the previousely mounted image for ID 1 has not been mounted, but the 3.5 disc drive has been moved from ID0 to ID 1 and has been mounted as ID 1
[image]

adtbm, 2019-11-25:
This behaviour is reproduceable and happens every time

adtbm, 2019-11-25:
UPDATE this BUG only happens when the image, you try to mount to ID0 is MEGA65.d81

adtbm, 2019-11-25:
Pauls updated (redesigned) Freezer from today (posted in Skype 12:51), solves the problem

closed 2021-02-09 by rdpeake