MEGA65/mega65-core #129: Bug reading files from real floppy disk directory

deftmega, 2019-11-08:
Around "the middle" of the directory, better end of 1st third, one or two files are not readable. you see them in the directory but get "file not found" if you try to load. or the load process just hangs and never ends.1st is more common. Anton: i can confirm this one ! it loads the file from a real 1581 ! but not from MEGA internal drive (from Skype chat)

gardners, 2019-11-09:
If you check which track/sector good and bad files start from, this would be great.  Basically any clues as to what is going on will be helpful.

gardners, 2020-05-29:
I have been investigating this, and the related problem with it being even less reliable on newer bitstreams.  There seems to be a few things going on:

1. The /DISKCHANGE line from the floppy doesn't seem to always occur.
2. When mounting a disk image or swapping from image to real floppy, there is no simulated /DISKCHANGE event, so DOS can get confused.
3. DOS can think the real floppy drive is on a different track than the one it is really on.  This is likely the cause of the read errors on some tracks, as it gets out of sync, and of course never knows to get back into sync.
4. If the drive hasn't gone to track 0, the head can be sitting funny, and not read properly (this shows up as very diffuse / random magnetix flux inversion intervals.

So, to fix it:
1. I am patching freeze menu disk chooser to always seek to track 0 when mounting a disk.
2. Adding a simulated /DISKCHANGE signal, that allows the dism image mounter to force a disk change event, even if the real drive doesn't support it.

But I still need to work out how and why the tracks get out of step. Are we stepping too quickly when using SD card? Because in theory at least, the drive head should always be matching the position of the SD card disk image from the DOS's perspective.

gardners, 2020-05-29:
Confirmed with floppytest utility from mega65-tools, that the drive ends up one track out sometimes.  We are not yet sure why.  To solve this once-for-all, we are implementing an auto-tuner for the drive controller, so that whether you have just switched to real floppy, or it has got out of sync otherwise, the drive will automatically step towards the track you are looking for. This relies on having a disk inserted that has the correct track numbers on its sectors, so some copy-protection schemes could confuse it badly.  But there aren't many for 1581 software, so I'm not expecting much trouble.

gardners, 2020-05-31:
I believe the auto-tuner has fixed this problem. The fix is staged on 138-hdmi-audio-27mhz (#138).

gardners, 2020-05-31:
The fix now requires independent confirmation. 
Use -b @138/47

@frehwagen can you please test under GEOS?

@deftmega can you please test or delegate testing loading files from further down on directory of 1581 disks? (the 1581 disks I have here are empty, so I can't do it myself, unless I get floppy writing working, which is another whole kettle of fish)?

Once both tests are in, we can advance to fix test confirmed.

deftmega, 2020-06-02:
Okay, I understood I should test issue 129 bitstream and came to the result, that it still hangs not finding one track, especially if I reset the MEGA65 while holding the MEGA key - because this does not reset the drive. When I rebooted into C65 mode and heard the drive reset, reading got better afterwards. So, after reading the above, I think that Paul is totally on the right track and will now head over to testing 138/47 and then report back here!

frehwagen, 2020-06-02:
Ok, testing #129, this is interesting, independent of @129/1 or @138/52:
I tested with 3 alternative drives as replace ment of the internal one. 2 of the alternatives work, while the original internal one and another one isn't.

Initially I was a bit mixup, because bit 4 of $D689 is diffently used in #217, which I already integrated into the currentl GEOS version.

gardners, 2020-06-02:
I'll try to reproduce the problem Deft reported.

Falk: To clarify, with this bitstream it works on all of your drives now?

frehwagen, 2020-06-03:
No, 2 drives work, the other 2 doesn't work, show the same problem I observed before.

gardners, 2020-06-03:
Update: I believe we have all agreed that auto-tune solves the first part of the problem, but that reading sectors can still get stuck, which we need to continue to investigate.

frehwagen, 2020-06-15:
I tested with GEOS and saw the same hanging when reading bytes for the blocks. Switching to DMA makes the floppy operation stable and reliable in GEOS, so I can confirm general disk reading work fine, but there is still something with the byte access.

gardners, 2020-06-15:
Thanks for the looking into this. My gut feeling is that there is a bit of a race condition with the stuffing and emptying of the floppy circular buffer, i.e., the handling of the buffer pointers and empty/full flags.  I guess for now use DMA in GEOS, and we should aim to create a simple test programme that can reproduce the problem, so that we can better characterise the problem and then confirm when we have a fix for this.

closed 2021-02-15 by rdpeake