MEGA65/mega65-core #133: Sprite can appear in front of right edge of left border

gardners, 2019-11-15:
As described, a sprite positioned over the edge of the left border appears in front of the right most column of pixels of the side border.
Confirmed to not be a problem with the right border.

gardners, 2019-11-15:
... but seemingly only the 2nd row of pixels in each logical raster of a V200 sprite.

gardners, 2019-11-15:
For V400 sprites, it is still every second physical raster line that does it.

gardners, 2019-11-15:
Problem seems specific to PAL video mode. No idea how or why this could be the case.

gardners, 2019-11-15:
Review after fixing #132 in case it magically fixes it.

lydon42, 2022-01-03:
It seems the problem has shifted to the left side (devel@8018e14).
[image]

lydon42, 2022-01-03:
This is still a thing with recent pre release development build.

ki-bo, 2024-09-01:
Is this still an issue or maybe even fixed by https://github.com/MEGA65/mega65-core/issues/815?