MEGA65/mega65-core #136: Suspect 1mhz branches get charged too many cycles sometimes

gardners, 2019-11-16:
Run wizball in PAL, and it will mostly work, but that the display flickerse with interrupts seeming to happen only every other frame, because something is going to slowly.  Disable charging for branches taken ($D7FB bit 0), and the problem goes away.

I have also seen indications elsewhere the branches might be incorrectly billed cycles, e.g., in synthmark64 results at 1mhz looking a bit low for things involving branches.

Needs to be investigated and fixed.

A temporary workaround is probably to just disable charging for these, as slightly too fast CPU shouldn't cause many problems compared with slightly too slow.

gardners, 2019-11-16:
Just as likely actually that badlines are getting triggered in the lower border.

gardners, 2019-11-16:
It looks like it might well be the triggering of badlines in the lower border that was the problem. In any case it has fixed it. There is still a funny glitch with the switching the vertical smooth scrolling from the score banner to the main playfield

gardners, 2019-11-16:
@frehwagen this one can also get merged in and closed when you are happy with it.  The remaining problems will become a separate issue.

lydon42, 2022-10-02:
From the commits this looks not like it was ever merged. Is this still a problem in recent cores? Who knows how to check?