MEGA65/mega65-core #138: Get HDMI audio output working

gardners, 2019-11-17:
Self-explanatory.

Depends on #114

gardners, 2019-11-17:
Begun implementing this.

Using the hdmidebug target, I am able to play a tone, but it is being  badly shaped, a the following images show:

[image]
[image]
[image]

The waveform is smooth, just badly enveloped.

Changing the HDMI initialisation frequency in vga_hdmi doesn't change the ~4Hz pulse train, so it is not some HDMI setup thing.

That the waveform is smooth suggests to me that it probably isn't samples being missed, but its hard to know for sure.

gardners, 2019-11-18:
New fascinating clue: HDMI video works with SPDIF audio enabled, only in NTSC mode: In PAL mode, the screen goes blank.  SPDIF audio is still very broken up in NTSC mode, so something is still not right.

gardners, 2019-11-18:
Found this nice reference: https://www.av-iq.com/avcat/images/documents/pdfs/digaudiochannelstatusbits.pdf

There should be 192 sample PAIRS per sub-frame, not 192 samples, according to this.  Reading more...

gardners, 2019-11-18:
But that might be wrong, as with that, there is not even an HDMI image.
What might also be an issue is if the parity bit is 1, then we need the inverted preamble.  It doesn't look like that ever actually gets sent.

gardners, 2019-11-18:
Another source also indicates 192 x L+R pairs: http://www.hardwarebook.info/S/PDIF
But finding good source material seems to be hard...

gardners, 2019-11-18:
Trying a different SPDIF implementation from OpenCores in case it solves the problem.

gardners, 2019-11-18:
So, still no picture with PAL mode when SPDIF audio is enabled.  But still picture NTSC mode, even if audio is enabled.  STill not convinced the audio is right. Will build full bitstream to test with games.

gardners, 2019-11-18:
Full bitstream: Still no picture in PAL. Picture and broken HDMI audio in NTSC.  Commando music reveals that it seems to be eliminating all high-frequencies.  That is, it sounds like only some samples are being accepted.  Note that this is with a totally different SPDIF implementation to before. This makes me think that something with the ADV7511 setup is wrong.

gardners, 2019-11-19:
See also commit 910ab4753ccd15754e3a0315c656bd80be01d266

gardners, 2019-11-19:
I forgot to tag that commit with this issue, but it might well be the fix: Set N and CTS values for the audio stream, so that proper clock recovery is possible by the receiver.  This at least made LOUD noise come out of the headphones with the test bitstream.  PAL video mode is still blank and silent.

gardners, 2019-11-19:
Build into full mega65r2 target bitstream, and music is now quite recognisable, but still badly distorted using earbuds in the HDMI monitor I have here.  HDMI TV still doesn't want to give a picture in either NTSC or PAL mode, but that might be HDMI interrupt handling, which I have reverted, and will try on a fresh bitstream when ready.

gardners, 2019-11-19:
I get the distinct impression that the problem is the N and CTS values need fine-tuning.  The best solution to this is to allow reading and writing of the I2C registers of the 7511 directly, as well as the automatic initialisation via the i2c_sender, that way I can fine tune the values without having to do repeated synthesis runs.

gardners, 2019-11-20:
Apparently it isn't unusual to have problems with Audio Clock Recovery (ACR) in HDMI: 

http://www.aes-media.org/sections/uk/Conf2011/Presentation_PDFs/14%20-%20john%20dawson%20-%20Audio%20Transport%20over%20HDMI%20-%20AES%202011.pdf

This also adds weight to the theory that getting the N and CTS values right is our best hope.  Certianly wrong values (like now) are going to hurt things.

Alternatively, we can increase the sample rate to 44100*(27.083333/27) and rely on the receiver to deal with the 0.31% difference as it would for any normal clock drift.  This would be 44237 samples per second in the SPDIF transmitter.

gardners, 2019-11-20:
The first option didn't work, so the above commits try the alternative of increasing the sample rate.

gardners, 2019-11-20:
That didn't work either. So let's take a step back, and implement a full R/W I2C memory mapped interface for the ADV7511, so that we can see what it thinks about the audio etc.
Will need to also add the i2c_sender functionality, so that it can automatically initialise the HDMI TX when required.

gardners, 2019-11-22:
Going slightly mad trying to get the memory-mapped I2C registers working.  Mapping of the existing M65R2 I2C registers is also being flaky, so suspecting the clock frequency bump might have pushed things out of tolerance, but need to investigate further.  Need a test harness for the I2C stuff, so that I can quickly make sure whether the hdmi_i2c stuff is actually correct or not.

gardners, 2019-11-22:
ok, with new HDMI I2C control and memory mapping of registers we are back to automatically configuring the HDMI display -- although it doesn't re-configure when monitor turns off or is disconnected, only on initial power up.  Writing to an ADV7511 register does trigger it, however.

gardners, 2019-11-22:
(The last commit above is a potential fix for this problem).

gardners, 2019-11-22:
It now mostly works, but the 7511 doesn't get powered back up after HDMI link loss. This means that the screen doesn't come back on automatically if monitor gets turned off and on.  Fix this by checking register $41 in our existing infinite read loop, and powering the chip back up if it has been powered off.

gardners, 2019-11-23:
Wrote HDMI status debug utility that lets me see things like N and CTS values.  This revealed that detected CTS was double the expected value, indicating we are generating samples at 88.2KHz, not 44.1KHz. Thus we need to double the value of N, and keep CTS the same.

gardners, 2019-11-23:
Well, that didn't fix it. But being able to play with registers in real-time revealed some things:
1. You have to turn HDMI audio TX off via reg $0A before you can change things like auto/manual CTS.
2. Switching to auto CTS got HDMI audio working very nicely in NTSC! (commits follow)

gardners, 2019-11-23:
Now to figure out why PAL video still shuts down when audio is enabled...

gardners, 2019-11-23:
Audio level is still WAY too high for HDMI audio. Mixer values need to be like $0707 or lower, so we are still at least 5 bits too loud.  Also some static in the sound sometimes (e.g., in parts of the krakout music).

gardners, 2019-11-23:
Static seems to be constant level, i.e., it sounds relatively louder if you reduce audio mixer gain, and quieter if you increase audio mixer gain.

gardners, 2019-11-23:
Not audible when there is no sound playing. Is it some harmonic of SID to HDMI audio output?

gardners, 2019-11-23:
For PAL mode, anything that ceases the audio stream restores video, including switching from SPDIF to another audio source, or telling the ADV7511 that the audio source uses external MCLK instead of recovered MCLK.

gardners, 2019-11-24:
Have gone through all ADV7511 I2C main map registers and confirmed that everything looks right in PAL vs NTSC modes, but PAL mode still shows no video when SPDIF audio is enabled. Running out of ideas as to what could be the cause.

gardners, 2019-11-25:
Found a possible cause: In PAL video data continues one raster line into the VSYNC period, which might be confusing the ADV7511.  Trying to fix this now...

gardners, 2019-11-27:
Some TVs/monitors have very broken audio, suggesting we aren't indicating the audio sample format correctly.

gardners, 2019-11-27:
Varying I2C register 14 doesn't change it.
Is it bigendian/littleendian perhaps?
Or just 192khz audio sample rate?

deftmega, 2020-03-17:
PAL over HDMI broken on HANNS-G 27" Monitor:
[image]

Disabling Audio completely makes it worse and adds flickering!

gardners, 2020-05-10:
Issue #165 has added perfect 27MHz pixel clock.  I'll see if that doesn't help us get HDMI audio working with the ADV7511 on all monitors.

gardners, 2020-05-31:
Compatibility test reports are being gathered in:

https://docs.google.com/spreadsheets/d/1Eg2ADZKcW-mgzo7DsoEzXYScAOobrWCLVbrtxMsdCtE/edit#gid=0

commit a51b5502 (2020-05-10): pull in more changes from HDMI mode / CPU timing fixes #138
closed 2021-02-15 by rdpeake