MEGA65/mega65-core #139: HDMI video briefly blinks off sometimes

gardners, 2019-11-17:
Very likely this is due to the way we update the HDMI I2C registers continuously.  Switching it to instead reconfigure whenever an interrupt is received from the ADV7511 should fix it.

gardners, 2019-11-22:
Various fixes in #138 may well fix this problem, or at least mitigate it. This includes interrupt driven automatic reenabling of the link, instead of blindly setting the registers 4 times a second.
The blinking could also be due to losing the HDMI link because of out of tolerance timing due to 27.083 vs 27.000 MHz pixel clock, since enabling the audio stream currently totally kills display in PAL mode.

gardners, 2019-12-08:
Haven't seen this issue for a while.  Will close 20DEC2019 if it is not seen again by then.

rdpeake, 2021-02-15:
Closing as no comment about it being seen since 20 december 2019

closed 2021-02-15 by rdpeake