MEGA65/mega65-core #140: During RESET, reset VIC-IV registers otherwise strange Sprite behaviour

adtbm, 2019-11-19:
sometimes, when the VIC-IV sprite tiling register hasn't been cleared after a reset, the sprites tile themself along the screen.

So whenever a reset is done, the VIC-IV tiling registers should be reset/cleared as well

adtbm, 2019-11-25:
[image]
[image]

gardners, 2019-11-26:
This will almost certainly be the sprite tiling register somehow getting set.  Probably an easy solution is to just force it clear under reset.

gardners, 2019-11-26:
The above should force clear the guilty register on each reset.

lydon42, 2024-08-17:
@adtbm Can you still recreate the bug? From which programs are the screenshots?

gardners, 2024-08-17:
The viciv.vhdl file should simply have code to reset all registers on reset. If it doesn't add too much to core size (which it shouldn't), then that's the solution.