MEGA65/mega65-core #141: Virtual VDC support is needed

bluewaysw, 2019-11-19:
To make MEGA65 compatible to run main GEOS 128 application in high resolution it is required to support a virtual VDC interface. We plan to simulate required VDC functionality by implementing related hypervisor traps. The FPGA implementation need to map the register access to related traps.

gardners, 2019-11-20:
You have a fair bit of this done already, right?

frehwagen, 2019-12-02:
Yes, it is about bringing it to kickstart/stable base, some cleanup and some more testing.
In general the VDC registers conflict with C65 uart registers, so and flag to enable VDC operation needs to be added as well.

gardners, 2019-12-03:
For the C65 UART register conflict, I'd make the traps only work in VIC-II IO mode, as no C128 GEOS app should need access to C65 registers.

gardners, 2019-12-10:
Above commits are first attempt to implement emulated VDC register and memory access. This happens in the CPU's memory re-mapping part.

gardners, 2020-01-04:
There was a problem with this causing the freeze menu to hang. Not sure of the actual cause. As a first defensive measure, I have made the VDC support so that it only works outside of hypervisor mode.

gardners, 2020-02-08:
Yes, both VDC and freeze menu should be working with this commit in place. Please let me know if it is good for you.

frehwagen, 2020-02-19:
After an initial implementation of VDC behaviour subset that is needed by some GEOS 128 application using trap and hypervisor services written in assembly we decided to try a small footprint version directly in vhdl. @gardners did the basic setup directly in the CPU's implementation and it tured out that this seems to be the best approach. We completed the vhdl implementation of the  subset functionality, it is ready for testing now.

@gardners proved that even if we use some mult, div and mod operations in the  vhdl code, there is acceptable impact on the logic complexity, and we can go with this. Even if the VDC registers conflict with the UART C65 register GEOS needs the registers available in VICIV mode, so we introduced an flag to specifically enable the VDC override.

frehwagen, 2020-02-19:
The register/bit to enable VDC simulation registers is $D710 bit 2. Defaults to 0 (disabled).

The simulated VDC ram is assumed to live at $40000. The current implementation assumes 640 pixel screen width (80 cards), mono color. This might be improved later to match VIC configuration better.

As of now the following VDC register are simulated:
$12/$13 - current address for byte read/write or bulk write
$1E - block count for bulk operation
$!F - read/write data byte or trigger bulk operation
$20/$21 - source address for bulk copy

The simulation is only active in non-hypervisor mode.

adtbm, 2020-02-19:
tested :
mega65r2-20191217.13-HEAD_n-a6ed3fb+DIRTY.bit (Issue #141 bitstream)

unfortunately not able to get geos to display:
**HDMI - NTSC**
[image]

Autoboot GEOS:
[image]

**HDMI - PAL**
[image]

Autoboot GEOS:
[image]

**VGA NTSC:**
[image]

**VGA PAL**
[image]

Freezermenu works with Thumbnails
[image]

gardners, 2020-02-20:
ok, weird. The "SCREEN NOT OPEN" error when trying to load the disk
directory says to me that there is something else weird going on there.
I'll try the GEOS here at some point and see what result I get.

Paul.

frehwagen, 2020-02-20:
True, sorry, I left a auto start program on this that is not yet complete at all. Maybe click to the mouse would move in to the desktop. The status of it was, I  trying to acces the SD card mount funktionality of the hypervisor. I wasn't remembering that it acutally is auto starting. I'll send an update D81 without this auto starter.

But all the latest bitstream I see are dated 20200108, not sure where yours came from.

adtbm, 2020-02-20:
i have updated my previous post, with the correct bitstream now.
the Autoboot, is working nicely, i have used your second version of the geos.d81

frehwagen, 2020-02-21:
Merged back into stable_base branch after successfully testing the bitstream and the VDC simulation function by running GeoWrite 128.

closed 2020-02-21 by frehwagen