MEGA65/mega65-core #144: White screen when going into BIOS

adtbm, 2019-11-25:
This Bug only happens after powering ON.
- hold ALT untill hypervisor submenu appears (To press 1 for utilities or 2 for SD tools)

gardners, 2019-11-26:
Please test with latest unstable bitstream, as I can't reproduce it here on N4DDR-widget board with head of unstable branch

gardners, 2020-02-25:
Do you still see this on the latest stable_base?

deftmega, 2020-03-20:
yes! Also entering FDISK is affected!
Happens on every Cold boot if you directly try to go to either FDISK or BIOS MENU!

gardners, 2020-03-24:
Problem is an interrupt of some kind triggering before the utility can mask them, combined with there being no valid ROM ready (even OpenROM requires the ROM vectors at $0300+ to be initialised)

gardners, 2020-03-25:
Fixed. Ready for merging in.

frehwagen, 2020-04-09:
Merged to stable_base/14.

closed 2020-04-09 by frehwagen