MEGA65/mega65-core #150: GEOS doesn't boot from IEC drive due to overzealous badline emulation

gardners, 2019-11-30:


frehwagen, 2019-11-30:
Additional info: The issues are not only in context of the booting geos, also just selecting/activating (at init) a serial drive in CONFIGURE causes issues with the serial communication to the drive. Because of this still there are some doubts that it relates to screen off behaviour of the bad lines :-)

gardners, 2019-11-30:
Do check it, as I expect all GEOS drive communications routines blank the screen to avoid badlines, since the timing would be way off, otherwise.  Let's see what it turns up.

gardners, 2019-12-02:
Should be fixed by commit 7988a7521.

@frehwagen can you please try with this commit and let me know if it fixes the problem.

Paul.

frehwagen, 2019-12-04:
By looking at the GEOS disk driver code it looks like it skips badlines when sending block bytes. Basically start raster line $32, $3A, $42... is skipped. From measuring the behaviour of M65 bad lines seems to be $34, $3C, $44....

gardners, 2019-12-05:
Good detective work! We will have to figure out how to solve this.  
Meanwhile, if you can try flipping bit 6 of $D05D (raster delay enable), to see if that makes the difference in badline rasters bigger or smaller, that would be great.
We might need to end up with a background colour buffer, so that we can match raster bars to the text display with equal delay.

frehwagen, 2019-12-11:
With the latest #141 branch bitstream (stable_base with hdmi video based), I get GEOS to boot from C64 mode screen, by
POKE53295,165:POKE53295,150:POKE53326,53
I think I'm running PAL50.
Initial PEEK53326 gives 57.

frehwagen, 2019-12-12:
Not sure anymore if the above is a stable observation.

gardners, 2020-08-17:
I'll take a look at the displaced badlines, as there is stil something funny with badlines, which also affects display in Boulderdash.
Meanwhile, I will make badlines adjustable upto 43 cycles each, since in most cases it is read instructions going on.  Not likely to be an issue for GEOS, as the drive code there just tries to avoid them.

lydon42, 2024-08-17:
Cleanup refresh: is this still an issue we need to track? @gardners @frehwagen

gardners, 2024-08-17:
The badline emulation in the VIC-IV is still quite problematic.  Boulderdash, for example shows this messing up with vertically stretched characters.  This is all interrelated. For GEOS, it should just set the CPU to 40MHz, and have the disk drive drivers updated to handle that (or better, the hardware IEC 4541, once it is ready, unless GEOS' fastloader is even faster than JD protocol).