MEGA65/mega65-core #152: HDMI display doesn't not always come on on power-up

gardners, 2019-12-08:
If an HDMI interrupt happens while initialising the HDMI chip, it will only cause the HDMI configuration to continue, instead of starting over from the beginning.  However, the interrupt may indicate that the HDMI TX chip has reset, and thus needs a full reset.

gardners, 2019-12-08:
To test, we need to do a large number of reset sequences in a row, and see if it happens.
Will quantify the probability under the bitstreams where it occurs.

gardners, 2019-12-08:
Seems to happen more often if the monitor has been asleep for a while.

gardners, 2019-12-08:
e.g., turn monitor off. Turn monitor on, and leave off long enough to go to sleep. 
But that also doesn't always cause it.

gardners, 2019-12-08:
The turn off M65, turn off monitor for long enough to go to sleep/power saving, and turn monitor back on, and finally turn on M65.
Do a series of runs:  Y = happened, - = means it didn't happen.
----Y----Y-YY

So it looks like the probability is >= 10%.  Twenty runs in a row without problem should be enough to confirm the problem is fixed.

gardners, 2019-12-08:
Also confirmed that it still happens without turning the screen off first.

gardners, 2019-12-08:
Seems to have fixed it.
Need to verify it, before retiring the issue.

adtbm, 2019-12-08:
No proper HDMI signal at all.

Let's start with VGA:
VGA looks nominal:
[image]

OK, main test HDMI:

i can confirm, that the Panel detects the HDMI signal immediately, but no proper picture. several attempts...
Examples:
[image]
[image]
[image]
[image]

adtbm, 2019-12-08:
sometimes, i receive a more or less...picture.
[image]

and even if the picture looks as bad as here:

[image]

The Freeze menu works always: Proof:

[image]

adtbm, 2019-12-08:
All results are with PAL. NTSC seems to work:

[image]

adtbm, 2019-12-08:
to summarize it:
The monitor gets the HDMI signal superfast, regardless of in PAL or NTSC !

But in PAL no picture, only freeze menu.
in NTSC picture OK !

gardners, 2019-12-09:
So I am suspecting that it could be HSYNC polarity that is a problem here.
In any case, it seems to be the HSYNC is being misinterpretted in some way.

Things to try:

sffd307c 0
sffd307c 40
sffd307c 80
sffd307c c0

Those four try all combinations of HSYNC and VSYNC polarity.

Then try disabling HDMI audio:

sffd361a 0

Each of the 8 combinations of the above would also be good to test: If it
is an older monitor, it might not do audio over HDMI well, and so the HDMI
audio could possibly cause a problem.

That's the main ideas I have for possible causes.

Paul.

rdpeake, 2021-02-15:
"ADV7511 fails HDMI 1.4a compliance tests at 720x576p and 720x480p in our configuration. Fixed on R3 PCB. Wont (and cant) fix on R2 PCB"

closed 2021-02-15 by rdpeake