MEGA65/mega65-core #156: Implement flashing of bitstreams into the QSPI flash

gardners, 2020-01-17:
Now that #155 , #153  and #149 are complete, we can implement writing new bitstreams to the QSPI flash.

gardners, 2020-02-13:
Now can write whole bitstreams, but sometimes doesn't succeed.
Need to add read-back after writing whole (as reading back after each write gets tripped up by a strange read buffer-like behaviour of the flash, as far as I can tell).
Need to implement QSPI writing, to speed up the writing process.
Need to speed up erase, by not repeatedly erasing each page by stepping only 512 forward during erase. We can step at least to the next 4KB boundary (this bug is probably related to the weird read-buffer thing).

gardners, 2020-02-13:
Flash erasing is now faster, but we are again having big problems reliably writing to the flash.
This is quite weird, as write_latch is being set etc.

gardners, 2020-02-13:
Will switch to QSPI for writing next, in case reading QSPI and writing SPI might do funny things.

gardners, 2020-02-28:
Reading registers via the ICAPE2 interface is proving a bit entertaining.  Output value, regardless of register read is $FFFFFF9B.  Tried adding a framework to do the switch to/from read/write, but still no luck, so far.

gardners, 2020-02-28:
CS/RW were switched. Testing with that fix...

gardners, 2020-02-28:
Hmm.. still no change.

gardners, 2020-02-28:
Turns out that you have to have started the FPGA from flash, not JTAG, just like when trying to reconfigure it. 

https://forums.xilinx.com/t5/Other-FPGA-Architecture/ICAPE2-Configuration-Readback-issue-Zynq-7Z010/td-p/782420

I now see something. Plumbing in register selection, to see if we can correctly determine the cycle on which to read.

gardners, 2020-02-28:
The above commits get us a long way towards running other cores, and running an upgraded MEGA65 core if one exists.

Basically, the Hypervisor checks if the FPGA has been reconfigured since power on. If not, then it will try to run the megaflash.prg program, that is pre-embedded in BRAM at $50000.
Even if the FPGA has been reconfigured since power on, if either joystick button or the TAB key is being held down, then it will try to run the megaflash.prg.
However, this happens only on first boot after each FPGA reconfiguration, since we cannot otherwise rely on the megaflash program being present in memory.
The megaflash program gets DMAd down to $0801, and JMP $080D starts it -- i.e., it runs in the Hypervisor memory context.  The Hypervisor writes the correct return address at $03F0/1 before jumping to $080D.  The megaflash.prg program will use this to return to the Hypervisor if it decides that it should neither load an alternate bitstream nor display the flash menu itself.

That's all the theory. There are bound to still be some errors in the bitstream built from the above, but it will hopefully be close enough that it won't take long to get it all working.

gardners, 2020-03-01:
These commits get us to the point where the flash menu and utility menu can be run on first boot, and the flash menu appears if TAB is held, but otherwise returns control immediately to HYPPO.

The main problem right now, is that our OpenROM doesn't work with this, and we have to use a real C65 ROM -- seemingly because the $E518 entry point is not compatible.  This might be because it messes with the MAP instruction, which then of course causes grief when the RTS occurs, and it then goes to lala land, because the memory map was changed.

gardners, 2020-03-01:
Created issue in open-roms repo: https://github.com/MEGA65/open-roms/issues/43

gardners, 2020-03-01:
Will also try a work around: Lock hypervisor memory map in place when in hypervisor mode.

frehwagen, 2020-03-17:
Using the current version of the #156 feature branch after switch on it seems to end in endless looping in 
dos_cluster_to_fat_sector
of hyppo/dos.

Initial tests show that the bootup passed flash_menu: in hyppo/main, but never end up at jmp 	
// Actually launch freeze menu
	jmp $080d
This in combination seems a bit strange, have to double check it.

frehwagen, 2020-03-20:
It seems strange that commit 02f2cc start with a flickering board, but latest commit be0ee5 doesn't flicker even if the assembly command order seem eqal?

frehwagen, 2020-03-30:
156- feature branch also come with changes for the following issues:

#149 - Cannot access QSPI configuration flash for updating bitstream under MEGA65 control
#155 - Incorporate fpgajtag functionality into monitor_load
#156 - Implement flashing of bitstreams into the QSPI flash
#153 - Work out how to do multiple bitstreams
#188 - Interrupt immediately following JMP ($xxxx) seems to save address of JMP, not its target
#144 - White screen when going into BIOS
#179 - MEGA65R1 target doesn't build
#167 - ROM checksum in Hyppo sums wrong region of memory
#188 - Interrupt immediately following JMP ($xxxx) seems to save address of JMP, not its target bug 
#189 - $D610 ASCII keyboard acceleration ignores CAPS LOCK key
#190 - Holding TAB on power-on not starting flash menu
#164 - MEGA65 R2 PCB external microSD card slot not working

adtbm, 2020-04-04:
Testing 156/61
used Bitstream:  mega65r2-20200401.11-HEAD_n-058b393+DIRTY.bit
i am not reporting any already known and reported bugs, that are not related to this test bitstream
and connected issues.

[#6](https://github.com/MEGA65/mega65-freezemenu/issues/6) in MEGA65-Freezemenu branch **SOLVED**  1 <-> 2 MHz frequency bug ! MEGA65, when starting in c64 mode is in 1MHz and freezer menu shows 1 MHz. 

#144  - **SOLVED** ! BIOS is working when powering on MEGA65 and holding ALT
#149  #153  #156  #190   - **SOLVED** ! able to access flash menu and to flash & erase Bitstreams/.COR,  NO SCROLL key works as flash menu hot-key always, as expected.

#189 - **NOT SOLVED** ! CAPS LOCK does not work in FDISK / writing problems, when using SHIFT LOCK in FDISK (

#164 partly solved. MEGA65 displays content of SD card, when two cards (SD & microSD) cards are inserted. it should display the content of the external (microSD). 

SUGGESTION: My opinion: if the MEGA65 would use always the internal (SD card) for MEGA65 relevant files (Freezer, etc.) and attaches the external microSD card as a kind of second diskdrive, with access to the d81 images as well would be even more versatile.

#155  #188 #179 #167  - can't state/test on that.

frehwagen, 2020-04-04:
Thanks for the details report, I additionally can prove #179 because there are resuling mega65r1 bitstream on filehost for @156/61.

adtbm, 2020-04-07:
Testing Results: 156/64
Bitstream: mega65r2-20200407.10-HEAD_n-42e315a+DIRTY.bit

i have also rechecked the issues from previous 156/61 bitstream test.

#176  **SOLVED** SD logic is correct, when both SD cards ( SD card & microSD card) are inserted, content of microSD card (external card) is accessible.

#159  **UPDATE** When entering the disc image chooser the first time, the directory is blank. leaving and re-entering solves this issue. 
The issue was not present in this bitstream:
unstabl 87a27d8+DIRTY,20191117.01
issue #159 updated - it is also valid for the R2 bitstream not NEXYS only.

#184 **SOLVED** you can choose a bitstream with cursor keys and it will load the selected bitstream
#185 **SOLVED** you can choose a bitstream with number keys and it will load the selected bitstream.
#189 **SOLVED** CAPS LOCK and SHIFT LOCK works both in FDISK menu.
#186 **SOLVED** CURSOR keys work reliably. in FDISK and disk image chooser, etc.
#190 **PARTLY SOLVED** NO SCROLL is the defined Flash menu hot key. REMAINDER Powering on with TAB pressed, still gives menu information popup window. not needed. NO SCROLL is key to enter flash menu. TAB is atm without any function. #190 updated
#197 **NOT SOLVED** more "strange" behaviours. #197 updated and [#199](https://github.com/MEGA65/mega65-core/issues/199) **CREATED**
#196 **NOT SOLVED** not able to type german letters. i've tried using ALT-key with all normal letters....(maybe the fault is me, not sure....)

#155 #188 #179 #167 #196  - can't state/test on those. these ones are still open and need to be tested.

gardners, 2020-04-07:
My test overnight.  Anton has tested most things more fully than I have, but here is a summary from me:

1. Bitstream starts fine with internal SD card.
2. ALT on first boot works to get into FDISK, and ASCII accelerated keyboard works fine in there, including ALT codes for Unicode Latin1 code page, typing quickly and other things that were problematic previously.
3. NO SCROLL on first boot gets flash menu, which has working ASCII accelerated keyboard behaviour, in particular, cursor UP not glitching.
4. Minor problem: TAB held on boot still shows a message, instead of being ignored. I would diff main.asm to unstable, as there are likely very few changes, and they will likely all be related to this.

frehwagen, 2020-04-07:
As for 4. there is no diff of main.asm that I consider logic relevant for TAB handling at startup. The diff hold 3 things:

- some differences in the sd mounting (not sure why those not has been picket to 156)
- hmdi audio related setup
- some string shortening regarding "NO SCROL"

adtbm, 2020-04-08:
When not having a microSD card inserted, MEGA65 doesn't boot.
MEGA65 is looking now for microSD as prime and ignoring normal SD card

frehwagen, 2020-04-09:
Done, merged to stable_base/14.

commit 2a7da342 (2020-03-15): Fix of last fix #156.
closed 2020-04-09 by frehwagen