MEGA65/mega65-core #157: Update to new Kick Assembler version

andrewoesten, 2020-01-22:
Kick Assembler releases >= v5.10 now contain new 65c02 instructions. Replace existing macros and migrate to them. This issue can be assigned to myself.

gardners, 2020-01-24:
Thanks! I have been underway relocating to Arkaroola, and will continue to have intermittent internet for the next 10 days or so.

lydon42, 2021-12-20:
Current mega65-core uses acme and ca65, I can't find any make rule that requires KickAss.
So this issue can be closed, and essentially KickAss could be removed from the mega65-core repo.

lgblgblgb, 2021-12-20:
My personal opinion only: it's not a good thing to always change assembler according to the latest fashion :) Just my personal experience, that Xemu used the Ophis format for some hypervisor debug symbol mode. Then mega65-core switched to acme for hyppo. Ok, now I altered things for acme (and I plan to use the "rep" file as well not only the "sym"). Then now switching to kickass, another change? :( Though for sure, I more speak about the assembler used for assembling hyppo.

titanlab, 2021-12-21:
I also think this issue can be closed. KickAss is also not open-source anymore.

lydon42, 2021-12-21:
I also discussed this with paul, we will remove KickAss from the core repo as it is not required anymore.

lydon42, 2021-12-21:
Removed in hwsc.

closed 2021-12-21 by lydon42