MEGA65/mega65-core #158: CPU and freeze glitching on recent bitstreams

gardners, 2020-02-07:
From C65 mode:

MONITOR
M6000

and it will display a couple of rows and hang.

Also, going into freeze menu and trying to resume barfs. 

It seems the CPU has problems with access memory around $6000.

Last build from 2019 is fine. First ones from 2020 are broken.  
Looking at git diff shows only VDC emulation and GeoRAM stuff changed in CPU.  So explore VDC emulation as probable cause.

gardners, 2020-02-07:
Last known working commit: 5ae744e 
First known non-working: e785e32

gardners, 2020-02-07:
Removing VDC emulation doesn't fix it. But removing both VDC and GeoRAM does.

gardners, 2020-02-07:
(Freeze menu then behaves correctly)

gardners, 2020-02-07:
Confirmed that it was the GeoRAM mapping that was the problem, i.e., the VDC code can be returned to use safely.

closed 2020-02-07 by gardners

bluewaysw, 2020-02-07:
Relates to this fix on the #141 branch:
https://github.com/MEGA65/mega65-core/commit/9b45fb11e035bba8a6973ae932430e65036c2130

gardners, 2020-02-18:
@frehwagen worth merging this onto stable_base once you are happy with it.